raspberrypi/cmprovision #3: Can no longer use rpiboot to get scriptexecute.img with Rev 5 CM4 HW

kileyhykawy, 2022-03-30:
There is a newer rev 5 version of CM4 hardware that requires RPi firmware newer than July 2021. The boot firmware in the `scriptexecute` folder is older than than (`VC_BUILD_ID_TIME` is `Jan 25 2021`). As such, the provisioning never starts.

kileyhykawy, 2022-03-30:
Verified that it works properly with rev 5 CM4 release in 1.3. Thanks!

closed 2022-03-30 by kileyhykawy

maxnet, 2022-07-05:
@kileyhykawy 

>Verified that it works properly with rev 5 CM4 release in 1.3.

Did you also test reprovisioning a CM4 module that already had existing content in eMMC by using the USB boot method, or did you only test provisioning fresh CM4 modules through network boot?

(Someone mentioned USB boot hanging for them on the forum, while network boot did work for them, and I was wondering if that may be a CM4 module revision specific issue as well, as I cannot reproduce it).