raspberrypi/cmprovision #5: I'm stuck trying to start using CM Provisioning

ASafwat-BioBusiness, 2022-07-24:
I've installed CM Provisioning, but need help to get started, I cannot see the connected CM4 devices on the web application, and don't know how to start.

JamesH65, 2022-07-24:
Check out this documentation https://pip.raspberrypi.com/categories/685/documents/RP-003468-WP/RP-003468-WP-2.pdf

ASafwat-BioBusiness, 2022-07-24:
> Check out this documentation https://pip.raspberrypi.com/categories/685/documents/RP-003468-WP/RP-003468-WP-2.pdf

Thank you very much.

ASafwat-BioBusiness, 2022-07-24:
@JamesH65, I'd like to share with you my attempt status:

1. I've build the required network setup
2. I can see the test CM4 unit on the web app
3. I've created an image from another CM4 unit with required environment and software (using win32 disk imager)
4. I've used 7Z Windows app to compress the .img file as .xz file
5. I've uploaded the image, created a project and marked as active
6. Project uses the uploaded image, required bootloader version, and required eeprom configurations

Now I cannot find how to start the provisioning process..
I can see in the dashboard page a log saying: "Could not provision, because there is no active project", this log is there from the first moment I've connected the test CM4 unit.

ASafwat-BioBusiness, 2022-07-24:
I've tried the USB boot (mentioned in 'Possible problems' section in the PDF), I attached the USB cable and put the jumper, the reconnected the power source to the test CM4 unit.

Still waiting for any UI change, but no luck.

I've explored the provisioning web app and found USB boot log, and seems it was waiting since it started early today, the a few minutes ago when I tried the USB boot, it seems it booted, but not sure about the result.

The problem is that the web app doesn't show any progress indication (pre-loader for example), so I'm not sure if there is any work in progress at the background.

USB Logs:

Jul 24 15:42:43 raspberrypi systemd[1]: Started rpiboot.
Jul 24 15:42:43 raspberrypi rpiboot[369]: RPIBOOT: build-date Jul 18 2022 version 20220718~085937 5a25e04b
Jul 24 15:42:43 raspberrypi rpiboot[369]: Loading: /var/lib/cmprovision/scriptexecute/bootcode.bin
Jul 24 15:42:43 raspberrypi rpiboot[369]: Waiting for BCM2835/6/7/2711...
Jul 24 20:44:59 raspberrypi rpiboot[369]: Loading embedded: bootcode4.bin
Jul 24 20:45:00 raspberrypi rpiboot[369]: Sending bootcode.bin
Jul 24 20:45:01 raspberrypi rpiboot[369]: Successful read 4 bytes
Jul 24 20:45:02 raspberrypi rpiboot[369]: Waiting for BCM2835/6/7/2711...
Jul 24 20:45:04 raspberrypi rpiboot[369]: Loading embedded: bootcode4.bin
Jul 24 20:45:05 raspberrypi rpiboot[369]: Second stage boot server
Jul 24 20:45:05 raspberrypi rpiboot[369]: Loading: /var/lib/cmprovision/scriptexecute/config.txt
Jul 24 20:45:05 raspberrypi rpiboot[369]: File read: config.txt
Jul 24 20:45:05 raspberrypi rpiboot[369]: Loading: /var/lib/cmprovision/scriptexecute/start4.elf
Jul 24 20:45:05 raspberrypi rpiboot[369]: File read: start4.elf
Jul 24 20:45:05 raspberrypi rpiboot[369]: Loading: /var/lib/cmprovision/scriptexecute/fixup4.dat
Jul 24 20:45:05 raspberrypi rpiboot[369]: File read: fixup4.dat
Jul 24 20:45:06 raspberrypi rpiboot[369]: Second stage boot server done
Jul 24 20:45:07 raspberrypi rpiboot[369]: Waiting for BCM2835/6/7/2711...
Jul 24 20:45:08 raspberrypi rpiboot[369]: Loading: /var/lib/cmprovision/scriptexecute/bootcode.bin
Jul 24 20:45:09 raspberrypi rpiboot[369]: Second stage boot server
Jul 24 20:45:09 raspberrypi rpiboot[369]: Loading: /var/lib/cmprovision/scriptexecute/config.txt
Jul 24 20:45:09 raspberrypi rpiboot[369]: File read: config.txt
Jul 24 20:45:09 raspberrypi rpiboot[369]: Loading: /var/lib/cmprovision/scriptexecute/config.txt
Jul 24 20:45:09 raspberrypi rpiboot[369]: File read: config.txt
Jul 24 20:45:09 raspberrypi rpiboot[369]: Loading: /var/lib/cmprovision/scriptexecute/scriptexecute.img
Jul 24 20:45:09 raspberrypi rpiboot[369]: File read: scriptexecute.img
Jul 24 20:45:12 raspberrypi rpiboot[369]: Loading: /var/lib/cmprovision/scriptexecute/bcm2711-rpi-cm4.dtb
Jul 24 20:45:12 raspberrypi rpiboot[369]: File read: bcm2711-rpi-cm4.dtb
Jul 24 20:45:12 raspberrypi rpiboot[369]: Loading: /var/lib/cmprovision/scriptexecute/config.txt
Jul 24 20:45:12 raspberrypi rpiboot[369]: File read: config.txt
Jul 24 20:45:12 raspberrypi rpiboot[369]: Loading: /var/lib/cmprovision/scriptexecute/overlays/dwc2.dtbo
Jul 24 20:45:12 raspberrypi rpiboot[369]: File read: overlays/dwc2.dtbo
Jul 24 20:45:13 raspberrypi rpiboot[369]: Loading: /var/lib/cmprovision/scriptexecute/overlays/spi-gpio40-45.dtbo
Jul 24 20:45:13 raspberrypi rpiboot[369]: File read: overlays/spi-gpio40-45.dtbo
Jul 24 20:45:13 raspberrypi rpiboot[369]: Loading: /var/lib/cmprovision/scriptexecute/cmdline.txt
Jul 24 20:45:13 raspberrypi rpiboot[369]: File read: cmdline.txt
Jul 24 20:45:13 raspberrypi rpiboot[369]: Loading: /var/lib/cmprovision/scriptexecute/kernel.img
Jul 24 20:45:13 raspberrypi rpiboot[369]: Loading: /var/lib/cmprovision/scriptexecute/kernel.img
Jul 24 20:45:13 raspberrypi rpiboot[369]: File read: kernel.img
Jul 24 20:45:17 raspberrypi rpiboot[369]: Second stage boot server done
Jul 24 20:45:18 raspberrypi rpiboot[369]: Waiting for BCM2835/6/7/2711...
Jul 24 20:45:18 raspberrypi rpiboot[369]: Loading: /var/lib/cmprovision/scriptexecute/bootcode.bin
Jul 24 20:45:19 raspberrypi rpiboot[369]: Second stage boot server
Jul 24 20:45:25 raspberrypi rpiboot[369]: Second stage boot server done
Jul 24 20:45:26 raspberrypi rpiboot[369]: Waiting for BCM2835/6/7/2711...

maxnet, 2022-07-24:
>The problem is that the web app doesn't show any progress indication (pre-loader for example)

It should show progress once there has been contact over Ethernet.
If for some reason that never happens, it is hard for server to guess what is going on.
You do can try connecting a HDMI display to the CMIO board to see if it gets stuck at booting (e.g. if display remains at rainbow pattern), or later (e.g. if something is wrong with Ethernet connectivity, such as STP).

ASafwat-BioBusiness, 2022-07-25:
I guess I need to restart the server CM to re-detect the connected units..

Now, I see on the dashboard the following error:
_**Image is not a valid disk image. Uncompressed size not dividable by sector size of 512 bytes**_

To recap, the image I'm using was created as follows:

1. I've created an image from another CM4 unit with required environment and software (using win32 disk imager)
2. I've used 7Z Windows app to compress the .img file as .xz file

maxnet, 2022-07-25:
>I guess I need to restart the server CM to re-detect the connected units..

Not the server.
Just power cycle the CMIO boards, to make them boot from network (or USB) again.

>Image is not a valid disk image. Uncompressed size not dividable by sector size of 512 bytes

Like the message says, disk drives (both of the spanning platter type and SD cards) always have a number of bytes that is dividable by 512 bytes.
If the image file is not that, there is something wrong with it.

>I've created an image from another CM4 unit with required environment and software (using win32 disk imager)

What is the exact file size of the image on your computer before compression?

maxnet, 2022-07-25:
BTW can you confirm you are using the **64-bit** edition of RPI OS on the CM server, and that running "uname -m" on the server confirms that?

If you accidentally are using 32-bit RPI OS that do can cause errors like this.
(PHP does not support integer values larger than 2^31 on 32-bit platforms, meaning you cannot reliably use image files over 2 GB.
When it encounters larger numbers PHP will start using floats instead of ints in which case you will get rounding errors, which will also cause the uncompressed file size number to be no longer dividable by 512).

ASafwat-BioBusiness, 2022-07-25:
> BTW can you confirm you are using the **64-bit** edition of RPI OS on the CM server, and that running "uname -m" on the server confirms that?

I'm using the 32-bit edition of RPI OS on the CM server :(

Also, the image I'm trying to provision itself is taken from a CM with 32-bit RPi OS.

> What is the exact file size of the image on your computer before compression?

15,634,268,160 bytes

> And what uncompressed file size does it show in the web interface in the "images" tab?

It's not displayed in bytes, it is displayed as follows:
Comp.: 1.9 GB
Uncomp.: 2.1 GB

maxnet, 2022-07-25:
>Uncomp.: 2.1 GB

Then there is probably a rounding error in the uncompressed value.
Its value is only used if you have "verify images" enabled in the project settings.
So you can disable that, or reinstall as 64-bit.

ASafwat-BioBusiness, 2022-07-25:
> Its value is only used if you have "Verify that image was written correctly" enabled in the project settings.

Yes, this option was enabled in the project settings.

> So you can disable that, or reinstall the CMprovision server as 64-bit.

I've reinstalled the CM Provision server as 64-bit, and repeated all steps and I'm finally done for a single CM4 unit :)

Now, I was trying to provision multiple CMs at a then same time.

I've connected 3 units of CM4 to the same switch of the sub network, but see nothing on the dashboard page and the CMs page.

I've checked the DHCP logs, below a sample of the logs:

Jul 25 21:09:39 raspberrypi-CM-Prvn-Server dnsmasq-tftp[521]: error 0 Early terminate received from 172.20.133.143
Jul 25 21:09:39 raspberrypi-CM-Prvn-Server dnsmasq-tftp[521]: failed sending /var/lib/cmprovision/scriptexecute/start4.elf to 172.20.133.143
Jul 25 21:09:40 raspberrypi-CM-Prvn-Server dnsmasq-tftp[521]: error 0 Early terminate received from 172.20.134.56
Jul 25 21:09:40 raspberrypi-CM-Prvn-Server dnsmasq-tftp[521]: failed sending /var/lib/cmprovision/scriptexecute/start4.elf to 172.20.134.56
Jul 25 21:10:06 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 27649999 available DHCP range: 172.20.0.2 -- 172.20.255.255
Jul 25 21:10:06 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 27649999 vendor class: PXEClient:Arch:00000:UNDI:002001
Jul 25 21:10:06 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 27649999 DHCPDISCOVER(eth0) e4:5f:01:94:25:0e
Jul 25 21:10:06 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 27649999 tags: client_is_a_pi, eth0
Jul 25 21:10:06 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 27649999 DHCPOFFER(eth0) 172.20.133.143 e4:5f:01:94:25:0e
Jul 25 21:10:06 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 27649999 requested options: 1:netmask, 3:router, 43:vendor-encap, 60:vendor-class,
Jul 25 21:10:06 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 27649999 requested options: 66:tftp-server, 67:bootfile-name, 128, 129,
Jul 25 21:10:06 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 27649999 requested options: 130, 131, 132, 133, 134, 135
Jul 25 21:10:06 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 27649999 next server: 172.20.0.1
Jul 25 21:10:06 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 27649999 sent size:  1 option: 53 message-type  2
Jul 25 21:10:06 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 27649999 sent size:  4 option: 54 server-identifier  172.20.0.1
Jul 25 21:10:06 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 27649999 sent size:  4 option: 51 lease-time  1h
Jul 25 21:10:06 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 27649999 sent size:  4 option: 58 T1  30m
Jul 25 21:10:06 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 27649999 sent size:  4 option: 59 T2  52m30s
Jul 25 21:10:06 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 27649999 sent size:  4 option:  1 netmask  255.255.0.0
Jul 25 21:10:06 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 27649999 sent size:  4 option: 28 broadcast  172.20.255.255
Jul 25 21:10:06 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 27649999 sent size:  4 option:  3 router  172.20.0.1
Jul 25 21:10:06 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 27649999 sent size:  9 option: 60 vendor-class  50:58:45:43:6c:69:65:6e:74
Jul 25 21:10:06 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 27649999 sent size: 17 option: 97 client-machine-id  00:52:50:69:34:40:31:c0:00:01:94:25:0e:6d...
Jul 25 21:10:06 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 27649999 sent size: 32 option: 43 vendor-encap  06:01:03:0a:04:00:50:58:45:09:14:00:00:11...
Jul 25 21:10:06 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 27649999 available DHCP range: 172.20.0.2 -- 172.20.255.255
Jul 25 21:10:06 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 27649999 vendor class: PXEClient:Arch:00000:UNDI:002001
Jul 25 21:10:06 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 27649999 DHCPREQUEST(eth0) 172.20.133.143 e4:5f:01:94:25:0e
Jul 25 21:10:06 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 27649999 tags: client_is_a_pi, eth0
Jul 25 21:10:06 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 27649999 DHCPACK(eth0) 172.20.133.143 e4:5f:01:94:25:0e
Jul 25 21:10:06 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 27649999 next server: 172.20.0.1
Jul 25 21:10:06 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 27649999 sent size:  1 option: 53 message-type  5
Jul 25 21:10:06 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 27649999 sent size:  4 option: 54 server-identifier  172.20.0.1
Jul 25 21:10:06 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 27649999 sent size:  4 option: 51 lease-time  1h
Jul 25 21:10:06 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 27649999 sent size:  4 option: 58 T1  30m
Jul 25 21:10:06 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 27649999 sent size:  4 option: 59 T2  52m30s
Jul 25 21:10:06 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 27649999 sent size:  4 option:  1 netmask  255.255.0.0
Jul 25 21:10:06 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 27649999 sent size:  4 option: 28 broadcast  172.20.255.255
Jul 25 21:10:06 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 27649999 sent size:  4 option:  3 router  172.20.0.1
Jul 25 21:10:06 raspberrypi-CM-Prvn-Server dnsmasq-tftp[521]: file /var/lib/cmprovision/scriptexecute/34a7f46d/start4.elf not found
Jul 25 21:10:06 raspberrypi-CM-Prvn-Server dnsmasq-tftp[521]: file /var/lib/cmprovision/scriptexecute/34a7f46d/start.elf not found
[... cut ...]
Jul 25 21:14:40 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 1830341133 available DHCP range: 172.20.0.2 -- 172.20.255.255
Jul 25 21:14:40 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 1830341133 vendor class: PXEClient:Arch:00000:UNDI:002001
Jul 25 21:14:40 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 1830341133 DHCPREQUEST(eth0) 172.20.134.56 e4:5f:01:94:24:f8
Jul 25 21:14:40 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 1830341133 tags: client_is_a_pi, eth0
Jul 25 21:14:40 raspberrypi-CM-Prvn-Server dnsmasq-dhcp[521]: 1830341133 DHCPACK(eth0) 172.20.134.56 e4:5f:01:94:24:f8

maxnet, 2022-07-26:
Can you attach a HDMI display to one of the CMIO boards and take a photo of the stage where it gets stuck?

The Ethernet switch connected to eth0 only has the CMprovision server and the CMIO boards connected, and no connection to another network or anything else attached, correct?

Also verify on the server that "sudo apt install tcpdump" "tcpdump -vvv -i eth0 stp" shows no STP packets.

ASafwat-BioBusiness, 2022-07-26:
> Can you attach a HDMI display to one of the CMIO boards and take a photo of the stage where it gets stuck?

Screen keeps flickering as it does when no OS.

> The Ethernet switch connected to eth0 only has the CMprovision server and the CMIO boards connected, and no connection to another network or anything else attached, correct?

Yes.

> Also verify on the server that "sudo apt install tcpdump" "sudo tcpdump -vvv -i eth0 stp" shows no STP packets.

I've run the command, it started to listen, and no packets there:

tcpdump: listening on eth0, link-type EN10MB (Ethernet), snapshot length 262144 bytes

maxnet, 2022-07-26:
>Screen keeps flickering as it does when no OS.

When no OS it should show a bootloader screen with a logo and some text. Not any of that?

ASafwat-BioBusiness, 2022-07-26:
I've figured out an issue with screen, that misled me for a while, fixed now.

But, while trying to solve screen problem I've disconnected one of the CMIOs.

I replaced the screen, reconnected the CMIO, then connected the HDMI cable to all boards one by one.

I found one of the CMs is up with the Linux image, and one is showing the below screen, and the third one (which I disconnected while testing screen) is showing a black screen!

I Returned back to the CM Provisioning web app and found two of the three CMs marked as provisioned!!

I restarted the three CMs and found another surprise:
The one that showed the screen in the above photo is still in the same state, and the one that showed the black screen showed a complete Linux image!!

As per timing on the CM Provisioning web app, it seems that the first image was provisioned while one of CMIOs is disconnected, and then after reconnecting the CMIO, it was provisioned, and the third one has failed.

I'm not really sure about the scenario!

maxnet, 2022-07-26:
>and the third one is showing the below screen:

That one seems to already have a half-installed operating system in eMMC.

In that case it will not boot from network.
Reprovision it using the USB boot method instead.

(- Disconnect power from that CMIO board.
- Connect cable between micro-USB "USB slave" port on CMIO board, and USB port on CMprovision server.
- Put jumper at "Fit jumper to disable eMMC boot" on CMIO board
- Reconnect power to CMIO board.)

ASafwat-BioBusiness, 2022-07-26:
Yes, I already done after sending you the last message, but I need to know:

1. Why the three CMs didn't provisioned yesterday and today until I disconnected one of them?
2. Why one of them has failed?

I can erase the three CMs and retry to provision again, but I need to know if there is an issue that prevented the server from detecting them together, it is very important to me.

maxnet, 2022-07-26:
>Why the three CMs didn't provisioned yesterday and today until I disconnected one of them?

No idea.
If a CMIO board fails to boot, you really need to have a display (or serial console cable) connected to figure out at what stage things go wrong.
So suggest you try to reproduce the issue while having a display connected...

Once the board boots, and checks-in with the CMprovision webserver it does should write to log when it starts provisioning, and if any errors occurs while writing the image, but we are unable to log anything that goes on before boot.

closed 2022-11-15 by maxnet