raspberrypi/cmprovision #6: Shutdown command request

aaronl1661, 2022-08-01:
At my company's assembly location, we require shutting down and turning on the server intermittently. I was wondering if it could be possible to add a shutdown command to gracefully shutdown the server without corrupting anything through the web application.

maxnet, 2022-08-01:
Not easier to add a physical button to the Pi playing server for that?
(Connect button between pin 5 & 6, add to /boot/config.txt: dtoverlay=gpio-shutdown )

aaronl1661, 2022-08-02:
We looked into that first, but the device we are using doesn't have easy access to physically add a button or easy access to GPIO pins. Before attempting that, we thought we would check to see if it was an easy feature to add or one that others may find useful. If it isn't likely to get implemented then we will revisit adding a button.

maxnet, 2022-08-02:
>Before attempting that, we thought we would check to see if it was an easy feature to add or one that others may find useful.

You can add it relatively easy yourself.

Not sure if it is suitable for inclusion as a general feature though, as others may not like the concept of adjusting security so that the webserver user can shutdown the server.

aaronl1661, 2022-08-02:
Thanks, we will look into these solutions.

closed 2023-03-21 by maxnet