raspberrypi/cmprovision #7: Access module connected to CM4

saeedjohar, 2022-08-18:
Hello,  

We have LTE modules connected to the CM4 over USB. Is it possible to access the LTE module to get the IMEI of the module during provisioning?

Thanks

maxnet, 2022-08-18:
Not out-of-the-box.
Would need to recompile the Linux kernel used with the kernel module that your LTE module uses.

Probably would need CONFIG_USB_ACM=y in the kernel config.
And a couple lines of shell script to send AT+GSN or a similar AT command to it, and read back the IMEI.

saeedjohar, 2022-08-19:
Okay then, I will look into Kernel modification and recompiling the kernel of the provisioner.
For the time being, I have a script that runs once when the device boots up for the first time and sends the mentioned data to a local server.

maxnet, 2022-11-15:
Added support for a couple common serial devices to the kernel configuration of scriptexecutor ( https://github.com/raspberrypi/scriptexecutor/commit/8047f15a46470539d6eeaa28422e16fc3a38353c ) bundled in the latest cmprovision.
You may want to try if your device is now recognized as well.

Closing this for now.

closed 2022-11-15 by maxnet

saeedjohar, 2024-05-29:
Hello @maxnet, 

Thank you for updating here. 
I have successfully tested this with CM4 IO carrier board but failed to load the USB on our two custom boards. 
The difference between these boards are; custom board uses LAN9514 as USB hub on one of them while the other uses TUSB4020BI. 

Do I need to alter anything on the scriptexecutor? 

Thank you

maxnet, 2024-05-29:
Don't think the USB HUB itself should matter, as I think they all speak the USB HUB class protocol thing.
But if something else is broken, e.g. the lines used for USB OTG negotiation, you may need another line in /var/lib/cmprovision/scriptexecute/config.txt
E.g. instead of "dtoverlay=dwc2" (USB OTG), force host mode with: "dtoverlay=dwc2,dr_mode=host"
Or remove line and use the XHCI thing with "otg_mode=1"

saeedjohar, 2024-05-30:
Adding `dr_mode=host` did the trick. 

Thank you