raspberrypi/cmprovision #9: Possibility to provision devices that have OS installed

JKalliomaki, 2022-12-21:
Hi!
It would be amazing for our possible use case that you could re-write the OS of a device that have OS already installed. Based on my testing of the program it was not possible, please correct me if I'm wrong.

maxnet, 2022-12-21:
Whether that is easily possible depends on your exact use case.

If you are just using cmprovision to write images, things are easy.
Just connect a USB<->micro USB cable between the cmprovision server, and the CMIO board's "USB slave" port.
And temporarily fit a jumper where it says "disable eMMC boot" on the CMIO board.
It will then boot the provisioning system through USB (rpiboot) instead of Ethernet, and that will also work if there is already an OS installed.
See also p. 14 of the whitepaper: https://pip.raspberrypi.com/categories/685-whitepapers-app-notes/documents/RP-003468-WP/Using-the-Compute-Module-Provisioner.pdf

Things get a lot more difficult if you have scripts configured in cmprovision that need to use USB ports during the provisioning, though.
Which will not work if the USB port is configured in device mode for the USB booting.

JKalliomaki, 2022-12-21:
Oh wow, I didn't know about that documentation. I think that can cover our use case well!

On another note, I think you could have link to the whitepaper in the README. I atleast found this tool from googling and ended up straight to this github page.

maxnet, 2022-12-21:
Good point. Added a link to the whitepaper.

closed 2022-12-21 by maxnet