raspberrypi/cmprovision #10: Use CM4 as provisioning server

mourix, 2023-01-31:
We are a tad hesitant to run a production process off an SD card. Is it possible for this software to run on a CM4 with EMMC and IO board? Or is there something that specifically requires the default Pi 4?

maxnet, 2023-01-31:
The .deb is platform independent so can install it on Intel/AMD gear as well.
Do need to use RPI OS for Intel as OS in that case, as we have one dependency (rpiboot package) that is not available on standard Debian/Ubuntu systems.
Can use higher end hardware like a SSD for storage (and 10 gigabit network connectivity, if you want to provision a lot of device simultanously) that way, if you have a need for that.

For network booting what you use as server does not matter.
If the boards you want to provision already have contents in eMMC, and you want to reinstall them, they no longer network boot though, and you need to use USB cables to USB boot them instead.
For that what USB controller you have on your server do can matter.
The 2 USB ports of a CMIO board would not be that great, if you use that as server, and need to connect more boards.
And I also have one AMD server that has issues with USB for some reason I have not figured out yet.
(things do not work on that server when USB cables that go from server to the devices being provisioned are already attached when Linux boots on server. With the kernel not enumerating USB correctly. But if I have them disconnected during boot and hotplug after boot everything works fine).
Using a Pi 4 as server does have the advantage that it is a known good hardware config...

ghollingworth, 2023-01-31:
It may be possible to use overlayfs in raspi-config to make the provisioner run from a read-only filesystem which would give you 100% reliability even with an SD card.

Of course, this would mean losing non-transient data (the database and log files).  @maxnet Would it be possible to store those somewhere more reliable (network file storage or external SSD)...

But once you introduce those, you may as well just boot a Pi 4 from SSD.

Gordon

maxnet, 2023-01-31:
>@maxnet Would it be possible to store 
>those somewhere more reliable (network file storage or external SSD)...

Technically yes.
But if you want to use an external USB SSD and do trust those, I would be tempted to just install RPI OS on that, and do not go for anything more complicated...

closed 2023-03-21 by maxnet