raspberrypi/cmprovision #11: Postinstall reboot script

thomas-at-rock7, 2023-02-10:
For my project a post install reboot script would be useful. I implemented a simple one:
```
#!/bin/sh
set -e
reboot -f
```
It executes successfully and newly flashed Pi reboots, however the CM Provision webserver still reports that Pi "Provisioning complete" as "No". So it appears this reboot command prevents CM Provisioning server confirming the Pi Provisioning is done. Perhaps the Pi reboots before the script exits successfully.

Is there a trick I am missing here? I have tried using a delay to the reboot (`reboot -f -d 5`) but that didn't work.

Thanks

thomas-at-rock7, 2023-02-17:
I missed the the "Run in backgound" checkbox. So ticking that with a slight tweak to the script solved it.
```
#!/bin/sh
set -e
sleep 30
reboot -f
```

closed 2023-02-17 by thomas-at-rock7

maxnet, 2023-02-27:
Another workaround would be to report completion to server from your post-installation script, like:

```
#!/bin/sh

# Upload post-installation log with script output
curl --retry 10 -g -F 'log=@/tmp/post.log' "http://${SERVER}/scriptexecute?serial=${SERIAL}&retcode=0&phase=postinstall"

# Mark as completed
TEMP=`vcgencmd measure_temp`
curl --retry 10 -g "http://${SERVER}/scriptexecute?serial=${SERIAL}&alldone=1&temp=${TEMP:5}

# Reboot
reboot -f
```

Problem with having a workflow that reboots do is that this will only work if your CM4 modules are blank, and you are able to let them perform a network boot.
Things get more complicated if you have a need to reprovision units that may already have contents in eMMC.
They will no longer network boot then, and when USB booting, if you reboot it will not boot to the OS you installed, but will USB boot again, and again, and again...

thomas-at-rock7, 2023-02-28:
Thanks for you're input. Our production workflow will only be using blank CM4 modules and we have other methods to flash CM4 modules with a OS installed. The reboot is needed on our end to begin first time boot process which involves time consuming firmware updates to peripheral devices.

maxnet, 2023-02-28:
>The reboot is needed on our end to begin first time boot process which involves time consuming firmware updates to peripheral devices.

Would personally rather do such firmware updates as pre-installation script instead.
So that if those fails for some reason (e.g. periperhal device broken), and the update script returns a non-zero exit code, it is properly registered, and the device will not be marked as provisioning complete.

thomas-at-rock7, 2023-02-28:
Sorry, little bit of confusion there. Ideally a pre-install script would be a better solution but the peripheral interfaces with the Pi serial ports and GPIO, so we need the OS running. These updates are done during the first boot of our OS, so as far as cmprovision is aware we have already completed and reported our postinstall status. Our OS does have steps to handle failed peripheral updates and the peripheral is always recoverable should it fail due to fallback firmware.