raspberrypi/cmprovision #13: Stuck after Link ready

saeedjohar, 2023-03-14:
Hello, 

To flash the Raspberry Pi Compute Module 4 (CM4), we are utilizing the cmprovision tool. We have successfully flashed thousands of CM4s with this tool, without any issues.

However, we have encountered problems with flashing new batches of CM4s. Despite this, we are still able to flash older versions of the CM4 without any problems.

In these new CM4, we are stuck at the following.
[image]

Not sure, how to proceed from here. 
Let me know if you need any logs. 

Thank you

maxnet, 2023-03-14:
Judging from your photo it seems the newer modules now have MAC addresses from a new range. (I must have missed the memo about that).

If you go the "settings" in the cmprovision webinterface, you should be able to add the new range to the MAC filter manually there.

d8:3a:dd:*:*:*

saeedjohar, 2023-03-14:
Oh great!
You are right. 

Thank you.

closed 2023-03-14 by saeedjohar

maxnet, 2023-03-15:
>I don't know if dnsmasq supports this but the 2711 firmware configures DHCP option97 by default with a GUID that 
>can be used to detect that the client is a Pi and the board-revision

Could probably be done in dnsmasq.conf with a construct among the lines of:

```
dhcp-match=set:client_is_a_pi,97,34:69:50:52
```

Would need to test that though.

>In earlier releases the client GUID (Option97) was just the serial number repeated 4 times. 

So we would still need the Mac filter as well to catch earlier firmware releases?

timg236, 2023-03-15:
DHCP OPTION97 was added in 2018 so it should be in all CM4 firmware, however, it would be worth leaving the MAC address filter in as well because if can be disabled and revert to the Pi3 behaviour where Option97 contains the serial number repeated 4 times.

maxnet, 2023-03-17:
>DHCP OPTION97 was added in 2018 so it should be in all CM4 firmware

Are you sure about that, because the changelog mentions March 2020 ?!
(Do not have access to the bootloader code, so cannot check myself)

```
2020-03-11 Add 2020-03-04 beta firmware recovery

* DHCP_OPTION97 - The default GUID has now changed to
  RPI4+BOARD_ID+ETH_MAC_LSB+SERIAL in order to make it easier to
  automatically identify Raspberry Pi computers. The old behaviour
  is enabled by setting DHCP_OPTION97=0 which simply repeats the serial
  number 4 times.
```

>however, it would be worth leaving the MAC address filter in as well because if can be disabled and revert to the Pi3 
>behaviour where Option97 contains the serial number repeated 4 times.

The annoyance I have with the MAC address filter is that is user editable within the web interface.
This was given as a requirement when the cmprovision software written, the idea being that the user could chose to continue running an old stable cmprovision version without having to upgrade and just add a new MAC range manually, when the need arises.

But it complicates things when the user does upgrade the cmprovision software.
Most users would just want newer boards to just work, so adding the new MAC range automatically on update would make sense.
But overriding configuration fields (the mac filter input field in the web interface) the user could have theoretically edited manually do is pretty bad practice.