raspberrypi/cmprovision #15: 11:30:24 Error during dd. Return code 1. Script output:

MattEyreKMO, 2023-03-15:
Hi all 

I am trying to provision a CM4 with a custom image and have uploaded the file as a .gz, plugged in the CM4 and recieved the following 

000 (0)	100000003cc4cabe	10:59:47 Error during dd. Return code 1. Script output:
000 (0)	100000003cc4cabe	10:50:05 Preinstall script complete. Starting to write image.
000 (0)	100000003cc4cabe	10:49:57 Provisioning started. Starting preinstall scripts.

I have checked the logs in the 'other settings' tab but nothing in there. I have also looked in the CM4 infomation and have the following 

===
Running pre-installation script 'Flash EEPROM firmware (stable/pieeprom-2022-04-26.bin)'
===
#!/bin/sh
set -e
curl --retry 10 --silent --show-error -g -o pieeprom.bin "http://172.20.0.1/uploads/pieeprom.bin"
echo "514e49703834474a9ac10317895ed1cbf72f01ee55ee8391bd659df383aa8b80  pieeprom.bin" | sha256sum -c
pieeprom.bin: OK
flashrom -p "linux_spi:dev=/dev/spidev0.0,spispeed=16000" -w "pieeprom.bin"
flashrom v1.1 on Linux 5.4.83-scriptexec (armv7l)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Found Winbond flash chip "W25X40" (512 kB, SPI) on linux_spi.
Reading old flash chip contents... done.

Any ideas what went wrong? I have tried two CM4s with seperate projects but this one produced an output on the CM4 informaiton. 

I am using a raspberry pi4 as the server and CM4 16gb and a dual gigabit carrier board foir the CM4

Firmware: stable/pieeprom-2022-04-26.bin

Thanks

Matt

maxnet, 2023-03-15:
Gives an error writing the image after writing for almost 10 minutes.

What is the exact uncompressed size in bytes of your custom image?
And how much space in bytes does you CM module have exactly? (Recall you can see that if you go to the "CMs" list and export as CSV).

[quote]
I have tried two CM4s with seperate projects but this one produced an output on the CM4 informaiton.
[/quote]

Can you rephrase that?
Not sure what you mean.
(Not a native English speaker, so be very clear with your words).

MattEyreKMO, 2023-03-15:
Hi 

Thanks for hte swift response :)

This is the listed size on the provisioner:

Comp.: 3.6 GB
Uncomp.: 31.9 GB
This is the size from the file (size on disk)

3.32 GB (3,571,449,856 bytes)

I have tried a 32 GB CM4 and also a 16GB (would the image have to be remade for a 16GB). Straight out of the box, unsure how I check this without an operating system?

or if you formatted the eMMC as pSLC, which reduces capacity by half 
(I didnt chose this option in the project settings)

This is the 32 gb CM4 attempt (which produced the information in the Pre-installation script output)
[export-cm-20230315 (1).csv](https://github.com/raspberrypi/cmprovision/files/10979960/export-cm-20230315.1.csv)

This is the 16 gb cm4 attempt (didnt report anything in the script output)

[export-cm-20230315 (2).csv](https://github.com/raspberrypi/cmprovision/files/10979967/export-cm-20230315.2.csv)

Any suggestions would be great 

Thanks

Matt

MattEyreKMO, 2023-03-15:
For clarity I am using the following board as a holder for the CM4
[image]

Configured like this without the boot and gnd connected 
[image]

maxnet, 2023-03-15:
Your CM4 that is sold as "32 GB" actually only has 31268536320 bytes of storage space.
I am afraid a 31.9 GB file system image is not going to fit in 31.2 GB space...

MattEyreKMO, 2023-03-15:
ahhh ok I need to resize the image to around 15gb so it will work on the 16gb versions too? The actual file size is 3.32gb and a copy from a 32gb card 

I will look at recreating it now

MattEyreKMO, 2023-03-15:
thanks so much

maxnet, 2023-03-15:
>ahhh ok I need to resize the image to around 15gb so it will work on the 16gb versions too?

Your 16 GB module has 15634268160 bytes space according to the CSV.
So as long as you make sure things are resized so that the end of the last partition is below that it should work.

MattEyreKMO, 2023-03-15:
brilliant thanks maxnet :) will look at doing this now

ghollingworth, 2023-03-15:
This might help

https://github.com/Drewsif/PiShrink/blob/master/pishrink.sh

maxnet, 2023-03-21:
[image]

In 1.6.2 it now should give a proper error message if image does not fit.