raspberrypi/cmprovision #16: Provisioning only work when I create a static IP for the MAC of the CM4 module

dougyip, 2023-03-28:
Using v1.6.2, provisioning doesn't proceed unless I assign a create a static IP address for the CM4 that I'm trying to provision. (for this case 172.20.0.2)  I create the static IP from the Other Settings - DHCP Server Settings page. Here's a snippet of dnsmasq log before and after I assign the static IP address:

-- Journal begins at Mon 2023-02-20 20:08:04 PST, ends at Tue 2023-03-28 12:17:02 PDT. --
...
Mar 28 11:57:24 RPI-Imager dnsmasq-dhcp[509]: 1592577310 requested options: 66:tftp-server, 67:bootfile-name, 128, 129,
Mar 28 11:57:24 RPI-Imager dnsmasq-dhcp[509]: 1592577310 requested options: 130, 131, 132, 133, 134, 135
Mar 28 11:57:24 RPI-Imager dnsmasq-dhcp[509]: 1592577310 next server: 172.20.0.1
Mar 28 11:57:24 RPI-Imager dnsmasq-dhcp[509]: 1592577310 sent size:  1 option: 53 message-type  2
Mar 28 11:57:24 RPI-Imager dnsmasq-dhcp[509]: 1592577310 sent size:  4 option: 54 server-identifier  172.20.0.1
Mar 28 11:57:24 RPI-Imager dnsmasq-dhcp[509]: 1592577310 sent size:  4 option: 51 lease-time  1h
Mar 28 11:57:24 RPI-Imager dnsmasq-dhcp[509]: 1592577310 sent size:  4 option: 58 T1  30m
Mar 28 11:57:24 RPI-Imager dnsmasq-dhcp[509]: 1592577310 sent size:  4 option: 59 T2  52m30s
Mar 28 11:57:24 RPI-Imager dnsmasq-dhcp[509]: 1592577310 sent size:  4 option:  1 netmask  255.255.0.0
Mar 28 11:57:24 RPI-Imager dnsmasq-dhcp[509]: 1592577310 sent size:  4 option: 28 broadcast  172.20.255.255
Mar 28 11:57:24 RPI-Imager dnsmasq-dhcp[509]: 1592577310 sent size:  4 option:  3 router  172.20.0.1
Mar 28 11:57:24 RPI-Imager dnsmasq-dhcp[509]: 1592577310 sent size:  9 option: 60 vendor-class  50:58:45:43:6c:69:65:6e:74
Mar 28 11:57:24 RPI-Imager dnsmasq-dhcp[509]: 1592577310 sent size: 17 option: 97 client-machine-id  00:34:69:50:52:41:31:a0:00:01:f8:de:b9:60...
Mar 28 11:57:24 RPI-Imager dnsmasq-dhcp[509]: 1592577310 sent size: 10 option: 43 vendor-encap  06:01:08:0a:04:00:50:58:45:ff
Mar 28 11:57:32 RPI-Imager dnsmasq-dhcp[509]: 1592577310 available DHCP range: 172.20.0.2 -- 172.20.255.255
Mar 28 11:57:32 RPI-Imager dnsmasq-dhcp[509]: 1592577310 vendor class: PXEClient:Arch:00000:UNDI:002001
Mar 28 11:57:32 RPI-Imager dnsmasq-dhcp[509]: 1592577310 DHCPDISCOVER(eth0) e4:5f:01:f8:de:b9
Mar 28 11:57:32 RPI-Imager dnsmasq-dhcp[509]: 1592577310 tags: eth0
Mar 28 11:57:32 RPI-Imager dnsmasq-dhcp[509]: 1592577310 DHCPOFFER(eth0) 172.20.38.21 e4:5f:01:f8:de:b9
Mar 28 11:57:32 RPI-Imager dnsmasq-dhcp[509]: 1592577310 requested options: 1:netmask, 3:router, 43:vendor-encap, 60:vendor-class,
Mar 28 11:57:32 RPI-Imager dnsmasq-dhcp[509]: 1592577310 requested options: 66:tftp-server, 67:bootfile-name, 128, 129,
Mar 28 11:57:32 RPI-Imager dnsmasq-dhcp[509]: 1592577310 requested options: 130, 131, 132, 133, 134, 135
Mar 28 11:57:32 RPI-Imager dnsmasq-dhcp[509]: 1592577310 next server: 172.20.0.1
Mar 28 11:57:32 RPI-Imager dnsmasq-dhcp[509]: 1592577310 sent size:  1 option: 53 message-type  2
Mar 28 11:57:32 RPI-Imager dnsmasq-dhcp[509]: 1592577310 sent size:  4 option: 54 server-identifier  172.20.0.1
Mar 28 11:57:32 RPI-Imager dnsmasq-dhcp[509]: 1592577310 sent size:  4 option: 51 lease-time  1h
Mar 28 11:57:32 RPI-Imager dnsmasq-dhcp[509]: 1592577310 sent size:  4 option: 58 T1  30m
Mar 28 11:57:32 RPI-Imager dnsmasq-dhcp[509]: 1592577310 sent size:  4 option: 59 T2  52m30s
Mar 28 11:57:32 RPI-Imager dnsmasq-dhcp[509]: 1592577310 sent size:  4 option:  1 netmask  255.255.0.0
Mar 28 11:57:32 RPI-Imager dnsmasq-dhcp[509]: 1592577310 sent size:  4 option: 28 broadcast  172.20.255.255
Mar 28 11:57:32 RPI-Imager dnsmasq-dhcp[509]: 1592577310 sent size:  4 option:  3 router  172.20.0.1
Mar 28 11:57:32 RPI-Imager dnsmasq-dhcp[509]: 1592577310 sent size:  9 option: 60 vendor-class  50:58:45:43:6c:69:65:6e:74
Mar 28 11:57:32 RPI-Imager dnsmasq-dhcp[509]: 1592577310 sent size: 17 option: 97 client-machine-id  00:34:69:50:52:41:31:a0:00:01:f8:de:b9:60...
Mar 28 11:57:32 RPI-Imager dnsmasq-dhcp[509]: 1592577310 sent size: 10 option: 43 vendor-encap  06:01:08:0a:04:00:50:58:45:ff
Mar 28 11:57:40 RPI-Imager dnsmasq[509]: exiting on receipt of SIGTERM
Mar 28 11:57:40 RPI-Imager systemd[1]: Stopping cmprovision-dnsmasq...
Mar 28 11:57:40 RPI-Imager systemd[1]: cmprovision-dnsmasq.service: Succeeded.
Mar 28 11:57:40 RPI-Imager systemd[1]: Stopped cmprovision-dnsmasq.
Mar 28 11:57:40 RPI-Imager systemd[1]: cmprovision-dnsmasq.service: Consumed 11.532s CPU time.
Mar 28 11:57:40 RPI-Imager systemd[1]: Starting cmprovision-dnsmasq...
Mar 28 11:57:40 RPI-Imager dnsmasq[7895]: dnsmasq: syntax check OK.
Mar 28 11:57:40 RPI-Imager dnsmasq[7898]: started, version 2.85 DNS disabled
Mar 28 11:57:40 RPI-Imager dnsmasq[7898]: compile time options: IPv6 GNU-getopt DBus no-UBus i18n IDN2 DHCP DHCPv6 no-Lua TFTP conntrack ipset auth cryptohash DNSSEC loop-detect inotify dumpfile
Mar 28 11:57:40 RPI-Imager dnsmasq-dhcp[7898]: DHCP, IP range 172.20.0.2 -- 172.20.255.255, lease time 1h
Mar 28 11:57:40 RPI-Imager dnsmasq-tftp[7898]: TFTP root is /var/lib/cmprovision/scriptexecute
Mar 28 11:57:40 RPI-Imager systemd[1]: Started cmprovision-dnsmasq.
Mar 28 11:57:48 RPI-Imager dnsmasq-dhcp[7898]: 1592577310 available DHCP range: 172.20.0.2 -- 172.20.255.255
Mar 28 11:57:48 RPI-Imager dnsmasq-dhcp[7898]: 1592577310 vendor class: PXEClient:Arch:00000:UNDI:002001
Mar 28 11:57:48 RPI-Imager dnsmasq-dhcp[7898]: 1592577310 DHCPDISCOVER(eth0) e4:5f:01:f8:de:b9
Mar 28 11:57:48 RPI-Imager dnsmasq-dhcp[7898]: 1592577310 tags: client_is_a_pi, known, eth0
Mar 28 11:57:48 RPI-Imager dnsmasq-dhcp[7898]: 1592577310 DHCPOFFER(eth0) 172.20.0.2 e4:5f:01:f8:de:b9
Mar 28 11:57:48 RPI-Imager dnsmasq-dhcp[7898]: 1592577310 requested options: 1:netmask, 3:router, 43:vendor-encap, 60:vendor-class,
Mar 28 11:57:48 RPI-Imager dnsmasq-dhcp[7898]: 1592577310 requested options: 66:tftp-server, 67:bootfile-name, 128, 129,
Mar 28 11:57:48 RPI-Imager dnsmasq-dhcp[7898]: 1592577310 requested options: 130, 131, 132, 133, 134, 135
Mar 28 11:57:48 RPI-Imager dnsmasq-dhcp[7898]: 1592577310 next server: 172.20.0.1
Mar 28 11:57:48 RPI-Imager dnsmasq-dhcp[7898]: 1592577310 sent size:  1 option: 53 message-type  2
Mar 28 11:57:48 RPI-Imager dnsmasq-dhcp[7898]: 1592577310 sent size:  4 option: 54 server-identifier  172.20.0.1
Mar 28 11:57:48 RPI-Imager dnsmasq-dhcp[7898]: 1592577310 sent size:  4 option: 51 lease-time  1h
Mar 28 11:57:48 RPI-Imager dnsmasq-dhcp[7898]: 1592577310 sent size:  4 option: 58 T1  30m
Mar 28 11:57:48 RPI-Imager dnsmasq-dhcp[7898]: 1592577310 sent size:  4 option: 59 T2  52m30s
Mar 28 11:57:48 RPI-Imager dnsmasq-dhcp[7898]: 1592577310 sent size:  4 option:  1 netmask  255.255.0.0
Mar 28 11:57:48 RPI-Imager dnsmasq-dhcp[7898]: 1592577310 sent size:  4 option: 28 broadcast  172.20.255.255
Mar 28 11:57:48 RPI-Imager dnsmasq-dhcp[7898]: 1592577310 sent size:  4 option:  3 router  172.20.0.1
Mar 28 11:57:48 RPI-Imager dnsmasq-dhcp[7898]: 1592577310 sent size:  9 option: 60 vendor-class  50:58:45:43:6c:69:65:6e:74
Mar 28 11:57:48 RPI-Imager dnsmasq-dhcp[7898]: 1592577310 sent size: 17 option: 97 client-machine-id  00:34:69:50:52:41:31:a0:00:01:f8:de:b9:60...
Mar 28 11:57:48 RPI-Imager dnsmasq-dhcp[7898]: 1592577310 sent size: 32 option: 43 vendor-encap  06:01:03:0a:04:00:50:58:45:09:14:00:00:11...
[... 75 lines cut ...]
Mar 28 11:58:08 RPI-Imager dnsmasq-dhcp[7898]: 2030322732 next server: 172.20.0.1
Mar 28 11:58:08 RPI-Imager dnsmasq-dhcp[7898]: 2030322732 sent size:  1 option: 53 message-type  2
Mar 28 11:58:08 RPI-Imager dnsmasq-dhcp[7898]: 2030322732 sent size:  4 option: 54 server-identifier  172.20.0.1
Mar 28 11:58:08 RPI-Imager dnsmasq-dhcp[7898]: 2030322732 sent size:  4 option: 51 lease-time  1h
Mar 28 11:58:08 RPI-Imager dnsmasq-dhcp[7898]: 2030322732 sent size:  4 option: 58 T1  30m
Mar 28 11:58:08 RPI-Imager dnsmasq-dhcp[7898]: 2030322732 sent size:  4 option: 59 T2  52m30s
Mar 28 11:58:08 RPI-Imager dnsmasq-dhcp[7898]: 2030322732 sent size:  4 option:  1 netmask  255.255.0.0
Mar 28 11:58:08 RPI-Imager dnsmasq-dhcp[7898]: 2030322732 sent size:  4 option: 28 broadcast  172.20.255.255
Mar 28 11:58:08 RPI-Imager dnsmasq-dhcp[7898]: 2030322732 sent size:  4 option:  3 router  172.20.0.1
Mar 28 11:58:08 RPI-Imager dnsmasq-dhcp[7898]: 2030322732 available DHCP range: 172.20.0.2 -- 172.20.255.255
Mar 28 11:58:08 RPI-Imager dnsmasq-dhcp[7898]: 2030322732 vendor class: udhcp 1.31.1
Mar 28 11:58:08 RPI-Imager dnsmasq-dhcp[7898]: 2030322732 DHCPREQUEST(eth0) 172.20.0.2 e4:5f:01:f8:de:b9
[... cut ...]
Mar 28 11:58:08 RPI-Imager dnsmasq-dhcp[7898]: 2030322732 sent size:  4 option: 58 T1  30m
Mar 28 11:58:08 RPI-Imager dnsmasq-dhcp[7898]: 2030322732 sent size:  4 option: 59 T2  52m30s
Mar 28 11:58:08 RPI-Imager dnsmasq-dhcp[7898]: 2030322732 sent size:  4 option:  1 netmask  255.255.0.0
Mar 28 11:58:08 RPI-Imager dnsmasq-dhcp[7898]: 2030322732 sent size:  4 option: 28 broadcast  172.20.255.255
Mar 28 11:58:08 RPI-Imager dnsmasq-dhcp[7898]: 2030322732 sent size:  4 option:  3 router  172.20.0.1

maxnet, 2023-03-28:
@doogyip
What EEPROM bootfirmware version does the CM4 module have?
(Shown on startup if it does not boot)

@timg236

When I network boot the single CM4 module I have for testing, I get a client-machine-id starting with 00:52:50:69:34
Which seems consistent with the documentation that says the GUID magic is "0x34695052 - *little endian*"

```
Mar 28 22:05:00 pi32 dnsmasq-dhcp[562]: 1092613896 DHCPDISCOVER(eth0) e4:5f:01:b7:9d:f1
Mar 28 22:05:00 pi32 dnsmasq-dhcp[562]: 1092613896 tags: client_is_a_pi, eth0
Mar 28 22:05:00 pi32 dnsmasq-dhcp[562]: 1092613896 DHCPOFFER(eth0) 172.20.230.5 e4:5f:01:b7:9d:f1
Mar 28 22:05:00 pi32 dnsmasq-dhcp[562]: 1092613896 requested options: 1:netmask, 3:router, 43:vendor-encap, 60:vendor-class,
Mar 28 22:05:00 pi32 dnsmasq-dhcp[562]: 1092613896 requested options: 66:tftp-server, 67:bootfile-name, 128, 129,
Mar 28 22:05:00 pi32 dnsmasq-dhcp[562]: 1092613896 requested options: 130, 131, 132, 133, 134, 135
Mar 28 22:05:00 pi32 dnsmasq-dhcp[562]: 1092613896 next server: 172.20.0.1
Mar 28 22:05:00 pi32 dnsmasq-dhcp[562]: 1092613896 sent size:  1 option: 53 message-type  2
Mar 28 22:05:00 pi32 dnsmasq-dhcp[562]: 1092613896 sent size:  4 option: 54 server-identifier  172.20.0.1
Mar 28 22:05:00 pi32 dnsmasq-dhcp[562]: 1092613896 sent size:  4 option: 51 lease-time  1h
Mar 28 22:05:00 pi32 dnsmasq-dhcp[562]: 1092613896 sent size:  4 option: 58 T1  30m
Mar 28 22:05:00 pi32 dnsmasq-dhcp[562]: 1092613896 sent size:  4 option: 59 T2  52m30s
Mar 28 22:05:00 pi32 dnsmasq-dhcp[562]: 1092613896 sent size:  4 option:  1 netmask  255.255.0.0
Mar 28 22:05:00 pi32 dnsmasq-dhcp[562]: 1092613896 sent size:  4 option: 28 broadcast  172.20.255.255
Mar 28 22:05:00 pi32 dnsmasq-dhcp[562]: 1092613896 sent size:  4 option:  3 router  172.20.0.1
Mar 28 22:05:00 pi32 dnsmasq-dhcp[562]: 1092613896 sent size:  9 option: 60 vendor-class  50:58:45:43:6c:69:65:6e:74
Mar 28 22:05:00 pi32 dnsmasq-dhcp[562]: 1092613896 sent size: 17 option: 97 client-machine-id  00:52:50:69:34:41:31:b0:00:01:b7:9d:f1:68...
Mar 28 22:05:00 pi32 dnsmasq-dhcp[562]: 1092613896 sent size: 32 option: 43 vendor-encap  06:01:03:0a:04:00:50:58:45:09:14:00:00:11...
```

However in Doogy's case it is:

```
Mar 28 11:57:24 RPI-Imager dnsmasq-dhcp[509]: 1592577310 sent size: 17 option: 97 client-machine-id  00:34:69:50:52:41:31:a0:00:01:f8:de:b9:60...
```

Could it be that there was a buggy EEPROM firmware version at some point that puts the magic in big endian instead of little endian?

Naturally, it's not really a problem to change my code to match on both...
But it do would be nice if such issues would be mentioned in the docs.

dougyip, 2023-03-28:
I flashed the eeprom with pieeprom-2023-01-11.bin

closed 2023-03-28 by maxnet

dougyip, 2023-03-29:
I still am having trouble provisioning with v1.6.3 with a CM4 with dynamic IP address. I did a clean install of 1.6.3.  Here's the log:

-- Journal begins at Mon 2023-02-20 20:08:04 PST, ends at Wed 2023-03-29 11:51:42 PDT. --
Mar 29 11:49:46 RPI-Imager systemd[1]: Starting cmprovision-dnsmasq...
Mar 29 11:49:47 RPI-Imager dnsmasq[491]: dnsmasq: syntax check OK.
Mar 29 11:49:47 RPI-Imager dnsmasq[512]: started, version 2.85 DNS disabled
Mar 29 11:49:47 RPI-Imager dnsmasq[512]: compile time options: IPv6 GNU-getopt DBus no-UBus i18n IDN2 DHCP DHCPv6 no-Lua TFTP conntrack ipset auth cryptohash DNSSEC loop-detect inotify dumpfile
Mar 29 11:49:47 RPI-Imager dnsmasq-dhcp[512]: DHCP, IP range 172.20.0.2 -- 172.20.255.255, lease time 1h
Mar 29 11:49:47 RPI-Imager dnsmasq-tftp[512]: TFTP root is /var/lib/cmprovision/scriptexecute
Mar 29 11:49:47 RPI-Imager systemd[1]: Started cmprovision-dnsmasq.
Mar 29 11:50:49 RPI-Imager dnsmasq-dhcp[512]: 2305295327 available DHCP range: 172.20.0.2 -- 172.20.255.255
Mar 29 11:50:49 RPI-Imager dnsmasq-dhcp[512]: 2305295327 vendor class: PXEClient:Arch:00000:UNDI:002001
Mar 29 11:50:49 RPI-Imager dnsmasq-dhcp[512]: 2305295327 DHCPDISCOVER(eth0) e4:5f:01:f8:de:90
Mar 29 11:50:49 RPI-Imager dnsmasq-dhcp[512]: 2305295327 tags: eth0
Mar 29 11:50:49 RPI-Imager dnsmasq-dhcp[512]: 2305295327 DHCPOFFER(eth0) 172.20.37.236 e4:5f:01:f8:de:90
Mar 29 11:50:49 RPI-Imager dnsmasq-dhcp[512]: 2305295327 requested options: 1:netmask, 3:router, 43:vendor-encap, 60:vendor-class,
Mar 29 11:50:49 RPI-Imager dnsmasq-dhcp[512]: 2305295327 requested options: 66:tftp-server, 67:bootfile-name, 128, 129,
Mar 29 11:50:49 RPI-Imager dnsmasq-dhcp[512]: 2305295327 requested options: 130, 131, 132, 133, 134, 135
Mar 29 11:50:49 RPI-Imager dnsmasq-dhcp[512]: 2305295327 next server: 172.20.0.1
Mar 29 11:50:49 RPI-Imager dnsmasq-dhcp[512]: 2305295327 sent size:  1 option: 53 message-type  2
Mar 29 11:50:49 RPI-Imager dnsmasq-dhcp[512]: 2305295327 sent size:  4 option: 54 server-identifier  172.20.0.1
Mar 29 11:50:49 RPI-Imager dnsmasq-dhcp[512]: 2305295327 sent size:  4 option: 51 lease-time  1h
Mar 29 11:50:49 RPI-Imager dnsmasq-dhcp[512]: 2305295327 sent size:  4 option: 58 T1  30m
Mar 29 11:50:49 RPI-Imager dnsmasq-dhcp[512]: 2305295327 sent size:  4 option: 59 T2  52m30s
Mar 29 11:50:49 RPI-Imager dnsmasq-dhcp[512]: 2305295327 sent size:  4 option:  1 netmask  255.255.0.0
Mar 29 11:50:49 RPI-Imager dnsmasq-dhcp[512]: 2305295327 sent size:  4 option: 28 broadcast  172.20.255.255
Mar 29 11:50:49 RPI-Imager dnsmasq-dhcp[512]: 2305295327 sent size:  4 option:  3 router  172.20.0.1
Mar 29 11:50:49 RPI-Imager dnsmasq-dhcp[512]: 2305295327 sent size:  9 option: 60 vendor-class  50:58:45:43:6c:69:65:6e:74
Mar 29 11:50:49 RPI-Imager dnsmasq-dhcp[512]: 2305295327 sent size: 17 option: 97 client-machine-id  00:34:69:50:52:41:31:a0:00:01:f8:de:90:89...
Mar 29 11:50:49 RPI-Imager dnsmasq-dhcp[512]: 2305295327 sent size: 10 option: 43 vendor-encap  06:01:08:0a:04:00:50:58:45:ff
Mar 29 11:51:05 RPI-Imager dnsmasq-dhcp[512]: 2305295327 available DHCP range: 172.20.0.2 -- 172.20.255.255
Mar 29 11:51:05 RPI-Imager dnsmasq-dhcp[512]: 2305295327 vendor class: PXEClient:Arch:00000:UNDI:002001
Mar 29 11:51:05 RPI-Imager dnsmasq-dhcp[512]: 2305295327 DHCPDISCOVER(eth0) e4:5f:01:f8:de:90
Mar 29 11:51:05 RPI-Imager dnsmasq-dhcp[512]: 2305295327 tags: eth0
Mar 29 11:51:05 RPI-Imager dnsmasq-dhcp[512]: 2305295327 DHCPOFFER(eth0) 172.20.37.236 e4:5f:01:f8:de:90
Mar 29 11:51:05 RPI-Imager dnsmasq-dhcp[512]: 2305295327 requested options: 1:netmask, 3:router, 43:vendor-encap, 60:vendor-class,
Mar 29 11:51:05 RPI-Imager dnsmasq-dhcp[512]: 2305295327 requested options: 66:tftp-server, 67:bootfile-name, 128, 129,
Mar 29 11:51:05 RPI-Imager dnsmasq-dhcp[512]: 2305295327 requested options: 130, 131, 132, 133, 134, 135
Mar 29 11:51:05 RPI-Imager dnsmasq-dhcp[512]: 2305295327 next server: 172.20.0.1
Mar 29 11:51:05 RPI-Imager dnsmasq-dhcp[512]: 2305295327 sent size:  1 option: 53 message-type  2
Mar 29 11:51:05 RPI-Imager dnsmasq-dhcp[512]: 2305295327 sent size:  4 option: 54 server-identifier  172.20.0.1
Mar 29 11:51:05 RPI-Imager dnsmasq-dhcp[512]: 2305295327 sent size:  4 option: 51 lease-time  1h
Mar 29 11:51:05 RPI-Imager dnsmasq-dhcp[512]: 2305295327 sent size:  4 option: 58 T1  30m
Mar 29 11:51:05 RPI-Imager dnsmasq-dhcp[512]: 2305295327 sent size:  4 option: 59 T2  52m30s
Mar 29 11:51:05 RPI-Imager dnsmasq-dhcp[512]: 2305295327 sent size:  4 option:  1 netmask  255.255.0.0
Mar 29 11:51:05 RPI-Imager dnsmasq-dhcp[512]: 2305295327 sent size:  4 option: 28 broadcast  172.20.255.255
Mar 29 11:51:05 RPI-Imager dnsmasq-dhcp[512]: 2305295327 sent size:  4 option:  3 router  172.20.0.1
Mar 29 11:51:05 RPI-Imager dnsmasq-dhcp[512]: 2305295327 sent size:  9 option: 60 vendor-class  50:58:45:43:6c:69:65:6e:74
Mar 29 11:51:05 RPI-Imager dnsmasq-dhcp[512]: 2305295327 sent size: 17 option: 97 client-machine-id  00:34:69:50:52:41:31:a0:00:01:f8:de:90:89...
Mar 29 11:51:05 RPI-Imager dnsmasq-dhcp[512]: 2305295327 sent size: 10 option: 43 vendor-encap  06:01:08:0a:04:00:50:58:45:ff
Mar 29 11:51:34 RPI-Imager dnsmasq-dhcp[512]: 2318325162 available DHCP range: 172.20.0.2 -- 172.20.255.255
Mar 29 11:51:34 RPI-Imager dnsmasq-dhcp[512]: 2318325162 vendor class: PXEClient:Arch:00000:UNDI:002001
Mar 29 11:51:34 RPI-Imager dnsmasq-dhcp[512]: 2318325162 DHCPDISCOVER(eth0) e4:5f:01:f8:de:90
[... cut ...]
Mar 29 11:51:42 RPI-Imager dnsmasq-dhcp[512]: 2318325162 sent size:  4 option: 28 broadcast  172.20.255.255
Mar 29 11:51:42 RPI-Imager dnsmasq-dhcp[512]: 2318325162 sent size:  4 option:  3 router  172.20.0.1
Mar 29 11:51:42 RPI-Imager dnsmasq-dhcp[512]: 2318325162 sent size:  9 option: 60 vendor-class  50:58:45:43:6c:69:65:6e:74
Mar 29 11:51:42 RPI-Imager dnsmasq-dhcp[512]: 2318325162 sent size: 17 option: 97 client-machine-id  00:34:69:50:52:41:31:a0:00:01:f8:de:90:89...
Mar 29 11:51:42 RPI-Imager dnsmasq-dhcp[512]: 2318325162 sent size: 10 option: 43 vendor-encap  06:01:08:0a:04:00:50:58:45:ff

maxnet, 2023-03-29:
If you do `cat /var/lib/cmprovision/etc/dnsmasq.conf` it does show two dhcp-match lines there?

```
dhcp-match=set:client_is_a_pi,97,0:52:50:69:34
dhcp-match=set:client_is_a_pi,97,0:34:69:50:52
```

(If not, your way of doing clean install, may not be clean enough).

dougyip, 2023-03-29:
Appears you’re right: my clean installed didn’t overwrite the dnsmasq.config properly.  The second dhcp-match=set:client_is_a_pi,97,0:34:69:50:52
Is missing.  I did an ‘apt remove’ of 1.6.2 then a ‘apt install’ of 1.6.3 but that doesn’t update the .config file.  Thanks for the help!

maxnet, 2023-03-29:
>I did an ‘apt remove’ of 1.6.2

'apt remove' only removes original program files.
'apt purge' also tries to get rid of configuration files that have been modified by the user (which happens if you added static IP entries through web interface).

dougyip, 2023-03-29:
Great tip, thanks!  I have confirmed the 1.6.3 is working for me once I added the missing line to dnsmasq.config Thanks again.

commit 4a719e0d (2023-03-28): Workaround for newer firmware sending opt 97 GUID big-endian

Closes #16