raspberrypi/cmprovision #17: CM4 is not listed / detected by server in the web-frontend

hobbit23, 2023-04-05:
Hi,
I have some problems with the Provisioning the Raspberry Pi Compute Module. The problem is right now, that I connected CM4 modules on a carrierboard via USB (alternative ETH) to a raspi4 (server) and do not see any CM4 modules via the webfrontend.

What I did so far:
- installed the Provisioning server on the raspi4 with the file cmprovision4_1.6.3_all.deb
- access to the webfrontend is succesfull
- uploading image is successfull
- check the CM4 modules on a windows PC with rpiboot and they can be detected
- contacted James Hughes several times for support (he leads me here)
- I tried to update the firmware via the webfrontend, its listed with the file pieeprom-2023-01-11.bin
- I do have access to the server via SSH

Can you please give me an advise or a teams call for checking the issue?
Happy to get a reply. Thanks in advance.
Nils

hobbit23, 2023-04-05:
this is my setup with USB connection
[image]

maxnet, 2023-04-05:
For CM4 the default configuration will assume it can use Ethernet after boot, even when USB was used to boot the device.
As most CM4 users only USB boot when network boot does not work for some reason (e.g. when there is already an operating system installed in the eMMC from a previous provisioning attempt, it will no longer PXE boot by default), but do have Ethernet connected.

You do should be able to change the default behavior by editing /var/lib/cmprovision/scriptexecute/config.txt
It currently looks like this:

```
# Set the pull ups for jumper reading
gpio=5,13,21=pu

initramfs scriptexecute.img

[cm4]
dtoverlay=dwc2
# For flashrom
dtparam=audio=off
dtoverlay=spi-gpio40-45

[pi3]
# Use USB over Ethernet on CM3
dtoverlay=dwc2,dr_mode=peripheral
cmdline=cmdline.txt.ipv6ll
```

And for a configuration that would always use USB for communication you would change it to look like this instead:

```
# Set the pull ups for jumper reading
gpio=5,13,21=pu

initramfs scriptexecute.img

# Use USB over Ethernet
dtoverlay=dwc2,dr_mode=peripheral
cmdline=cmdline.txt.ipv6ll

[cm4]
# For flashrom
dtparam=audio=off
dtoverlay=spi-gpio40-45
```

hobbit23, 2023-04-05:
I do not understand you completely. 
For setting the configuration as mentioned, I need to connect to the new CM4 which is not yet flashed. 
OR
do you mean to set the configuration in the server files?

hobbit23, 2023-04-05:
"For CM4 the default configuration will assume it can use Ethernet after boot, even when USB was used to boot the device."
what do you mean?
CM4 can boot via Ethernet after power on? this does not match with "when USB was used to boot the device"

maxnet, 2023-04-05:
>do you mean to set the configuration in the server files?

Yes, /var/lib/cmprovision/scriptexecute/config.txt is a file on the cmprovision server.
You can edit it using any text editor you have available on the server.
Do make sure you run the editor with sudo to be able to save your changes to the file.

E.g. you can open a terminal window and type "sudo mousepad /var/lib/cmprovision/scriptexecute/config.txt" if you are running the full edition of RPI OS on the server that has GUI.
Or you can use "sudo nano /var/lib/cmprovision/scriptexecute/config.txt" if you are running RPI OS Lite on the server.

>CM4 can boot via Ethernet after power on?

Yes, CM4 modules network boot over Ethernet if they are fresh out of the factory.
That is how cmprovision is normally used. Normally you do NOT use USB.

Suggest you read the manual if you have not done so: https://pip.raspberrypi.com/categories/685-whitepapers-app-notes/documents/RP-003468-WP/Using-the-Compute-Module-Provisioner.pdf

When using CM4 modules, USB boot is normally only an emergency procedure to use if your module no longer network boots for some reason. This procedure is described on page 13 of the manual ("eMMC already programmed")
After you managed to boot the board, Ethernet is used for normal communication with the server by default.

But like I said, if you want to use USB for further communication as well, that do is possible by editing  /var/lib/cmprovision/scriptexecute/config.txt
I would only use that if you really cannot use Ethernet though. E.g. if you want to use in-system-programming and your  custom board design does not have an Ethernet port.

hobbit23, 2023-04-05:
yes thank you, I read the manual already.
I also want to connect and flash via ETH but as it is not working i tried via USB

maybe I have to switch to USB later as we have to get rid of the ETH connector. Anyway, one of those options have to work soon.

So you advise me to power the board via USB and have the ethernet connection to the server after modification of the config.txt?

hobbit23, 2023-04-05:
how should the config.txt must look like to have a boot option via ETH?

maxnet, 2023-04-05:
>So you advise me to power the board via USB and have the ethernet connection to the server after modification of the 
>config.txt?

You only have to modify it if you want to use USB for everything, and do NOT want to use Ethernet.

- Do NOT modify it if you want to network boot over Ethernet. It already does that by default, if you connect only Ethernet, and have the "boot" switch on your Waveshare board set to "off"
- Do NOT modify it if you want to USB boot, but use Ethernet after boot. It already does that by default, if you connect both USB and Ethernet, and have the "boot" switch on your Waveshare board set to "on".

Be aware that for network booting over Ethernet:
- the CM4 module must be fresh. Once you installed an operating system on the eMMC of the module, it will no longer network boot.
- it can take 30 seconds or so before it network boots, because it tries other boot methods first.

hobbit23, 2023-04-05:
"Do NOT modify it if you want to network boot over Ethernet. It already does that by default, if you connect only Ethernet, and have the "boot" switch on your Waveshare board set to "off""

this is not working right now. the CM4 is fresh and I already waited quite long time

maxnet, 2023-04-05:
>this is not working right now. the CM4 is fresh and I already waited quite long time

Can you connect the HDMI of the CM4 board to a display, and post a photo of the bootloader screen you get?

hobbit23, 2023-04-05:
"Can you connect the HDMI of the CM4 board to a display, and post a photo of the bootloader screen you get?"

yes i can but there is only "the end" of the booting sequence of the server. I can log in to the server via ssh after that

yes i am using a standard ETH cable and already changed in between to another.

maxnet, 2023-04-05:
>yes i can but there is only "the end" of the booting sequence of the server. 

CM4 board = your Waveshare board
Connect HDMI to that, not the server.

hobbit23, 2023-04-05:
[image]

hobbit23, 2023-04-05:
the server is logging this

[image]

maxnet, 2023-04-05:
```
DHCP packet received on eth0 which has no address.
```

Do you have an Ethernet switch you can put between the Pi and the CM4?
Think the server gets confused because you are using a direct cable between the Pi and CM4, and it responds badly to the network link carrier going up and down all the time.

If you really do not have an Ethernet switch handy, you can also try if it works better if add to /etc/dhcpcd.conf on the server:

```
nolink
```

Reboot server after that.

maxnet, 2023-04-13:
- If you followed the instructions eth0 on the server should have IP 172.20.0.1 and a subnet of 255.255.0.0 (cidr /16).
Double check with "ip addr" that it shows 172.20.0.1/16
Our dnsmasq configuration will not hand out IPs if you configured an IP-address from any other range.

- You do are using latest cmprovision? Double check /var/lib/cmprovision/etc/dnsmasq.conf shows this:

```
# No DNS
port=0

# tftp
enable-tftp
tftp-root=/var/lib/cmprovision/scriptexecute

# dhcp
dhcp-match=set:client_is_a_pi,97,0:52:50:69:34
dhcp-match=set:client_is_a_pi,97,0:34:69:50:52

log-dhcp
dhcp-range=172.20.0.2,172.20.255.255,255.255.0.0
pxe-service=tag:client_is_a_pi,0,"Raspberry Pi Boot"
dhcp-leasefile=/var/lib/cmprovision/etc/dnsmasq.leases
no-ping
```

In particular that there is "dhcp-range=172.20.0.2,172.20.255.255,255.255.0.0" without any reference to tags.

hobbit23, 2023-04-13:
"You do are using latest cmprovision? Double check /var/lib/cmprovision/etc/dnsmasq.conf shows this:"

i am using the latest version. I always check this when working on that

hobbit23, 2023-04-13:
i checked the file dnsmasq.conf and have the same then you wrote before

hobbit23, 2023-04-13:
I also deleted "nolink"

hobbit23, 2023-04-13:
172.20.0.1 not reachable with direct connection. I have configured my ethernet port like this
[image]

maxnet, 2023-04-13:
That does NOT look good.
A 169.254.x.x auto configuration address is set if you do NOT have a static IP-address configured, and it cannot find a DHCP server.

Make sure you have in /etc/dhcpcd.conf:

```
interface eth0
static ip_address=172.20.0.1/16  
```

And reboot afterwards.

maxnet, 2023-04-13:
>what about the other lines with 192....?

Please follow instructions in the manual to the letter, and ignore all other lines in /etc/dhcpcd.conf.
They are COMMENTED out EXAMPLES of things you do can with dhcpcd.

We do not want a default gateway, nor IPv6, nor anything else.
Just 172.20.0.1/16 on eth0.

hobbit23, 2023-04-13:
okay. so I delete the # so it looks like this.
---
# A sample configuration for dhcpcd.
# See dhcpcd.conf(5) for details.

# Allow users of this group to interact with dhcpcd via the control socket.
#controlgroup wheel

# Inform the DHCP server of our hostname for DDNS.
hostname

# Use the hardware address of the interface for the Client ID.
clientid
# or
# Use the same DUID + IAID as set in DHCPv6 for DHCPv4 ClientID as per RFC4361.
# Some non-RFC compliant DHCP servers do not reply with this set.
# In this case, comment out duid and enable clientid above.
#duid

# Persist interface configuration when dhcpcd exits.
persistent

# Rapid commit support.
# Safe to enable by default because it requires the equivalent option set
# on the server to actually work.
option rapid_commit

# A list of options to request from the DHCP server.
option domain_name_servers, domain_name, domain_search, host_name
option classless_static_routes
# Respect the network MTU. This is applied to DHCP routes.
option interface_mtu

# Most distributions have NTP support.
#option ntp_servers

# A ServerID is required by RFC2131.
require dhcp_server_identifier

# Generate SLAAC address using the Hardware Address of the interface
#slaac hwaddr
# OR generate Stable Private IPv6 Addresses based from the DUID
slaac hwaddr

# Example static IP configuration:
interface eth0
static ip_address=172.20.0.1/16
#static ip6_address=fd51:42f8:caae:d92e::ff/64
#static routers=192.168.0.1
#static domain_name_servers=192.168.0.1 8.8.8.8 fd51:42f8:caae:d92e::1

# It is possible to fall back to a static IP if DHCP fails:
# define static profile
#profile static_eth0
#static ip_address=192.168.1.23/24
#static routers=192.168.1.1
#static domain_name_servers=192.168.1.1

# fallback to static profile on eth0
#interface eth0
#fallback static_eth0
---

by the way: this is not part of the instructions in the manual :/

maxnet, 2023-04-13:
>by the way: this is not part of the instructions in the manual :/

???

Page number 6, the very first step:

[image]

Reboot afterwards.

Check with "ip addr"
(Checking is not part of the manual, but it does not hurt if you are experiencing problems).

hobbit23, 2023-04-13:
ah great, now I can see the device.
[image]

how can I start the project to flash the CM4?

hobbit23, 2023-04-13:
> > by the way: this is not part of the instructions in the manual :/
> 
> ???
> 
> Page number 6, the very first step:
> 
> [image]
> 
> Reboot afterwards.
> 
> Check with "ip addr" (Checking is not part of the manual, but it does not hurt if you are experiencing problems).

yes, I did this at the beginning. maybe it is not working together with the wifi

maxnet, 2023-04-13:
>how can I start the project to flash the CM4?

See page 7, 8 and 9 of manual.
Report back if you have more specific questions.

hobbit23, 2023-04-13:
sorry to ask again. The Manual page 7,8,9 i already read and proceed. but how can i start the flashing process after uploading an image, setting a project and now see the CM4 listed?

thank you for helping

hobbit23, 2023-04-13:
great, thank you! now it is working :-) thanks a lot

maxnet, 2023-04-13:
Good to hear it is solved now.
Open a new issue, if you come across any other problems.

closed 2023-04-13 by maxnet