raspberrypi/cmprovision #18: Enabling i2c using provisioning script

philbegg, 2023-04-06:
I am currently using 
```
#!/bin/sh
set -e
echo 'i2c-dev' >> /etc/modules
```

as a script in the cm4 provision tool to try and enable i2c. When I boot the CM4 the /etc/modules file does not have i2c-dev appended to it.

What am I missing?

maxnet, 2023-04-06:
>What am I missing?

That you are editing the /etc/modules of the utility operating system that is network booted to do the provisioning, and not the operating system that lives on eMMC.

You typically want something like this instead (untested):

```
#!/bin/sh
set -e
mkdir -p /mnt/root
mount -t ext4 $PART2 /mnt/root
echo 'i2c-dev' >> /mnt/root/etc/modules
umount /mnt/root
```

That do is assuming it is a plain normal RPI OS image that has one FAT partition and one ext4 partition.
Things get more complicated with A-B setups, and operating systems that install themselves on first boot (like Ubuntu Core).

philbegg, 2023-04-11:
Thanks for the help - it is RPI OS so this worked great thanks.

closed 2023-04-11 by philbegg