raspberrypi/cmprovision #20: server is starting again and again flashing CM4

hobbit23, 2023-04-14:
The server is set up but in some cases, it starts all o ver again the new flasing process. 
[image]
Any idea about this?

maxnet, 2023-04-14:
Could be a power problem.
Make sure you power the modules separately and not off Pi's USB port.

hobbit23, 2023-04-14:
with an external power supply the board are not visible by the server. I expect, that the rpiboot is not possible then even the should do it via ETH. I tried it with different power supply plugs. What do you recommend?

maxnet, 2023-04-14:
>with an external power supply the board are not visible by the server. 

If they units are already flashed, you do indeed need to connect them by USB to the server.
But do not do so directly, but through a **powered** USB hub that has an adapter.

(Or program the units with normal CMIO boards instead of third-party boards. The official boards have separate barell connector for power.)

maxnet, 2023-04-14:
>even the "fresh" CM4 modules can not be connected by the server

Make sure the boot switch of fresh modules is set to "eMMC boot" "ON" so that they can network boot.

hobbit23, 2023-04-14:
yes I tried this but no effort

hobbit23, 2023-04-14:
server is starting again and again flashing two CM4. 
in dmesg i can see this:

  +0.000018] usb 1-1.1: Product: BCM2711 Boot
[  +0.000015] usb 1-1.1: Manufacturer: Broadcom
[  +0.000014] usb 1-1.1: SerialNumber: 74b0ffaa
[  +2.392993] usb 1-1.1: USB disconnect, device number 48
[  +0.226026] usb 1-1.1: new high-speed USB device number 49 using xhci_hcd
[  +1.135990] usb 1-1.1: new full-speed USB device number 50 using xhci_hcd
[  +0.102107] usb 1-1.1: not running at top speed; connect to a high speed hub
[  +0.008001] usb 1-1.1: New USB device found, idVendor=0a5c, idProduct=2764, bcdDevice= 0.00
[  +0.000029] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=1
[  +0.000015] usb 1-1.1: Product: BCM2711 Boot
[  +0.000012] usb 1-1.1: Manufacturer: Broadcom
[  +0.000012] usb 1-1.1: SerialNumber: Broadcom
[ +15.938219] usb 1-1.1: USB disconnect, device number 50
[  +0.479730] usb 1-1.1: new high-speed USB device number 51 using xhci_hcd
[  +5.296193] usb 1-1.1: device descriptor read/64, error -110
[  +8.884013] usb 1-1.1: device descriptor read/64, error -71
[  +0.187751] usb 1-1.1: new high-speed USB device number 52 using xhci_hcd
[  +0.100468] usb 1-1.1: config index 0 descriptor too short (expected 55, got 32)
[  +0.000496] usb 1-1.1: New USB device found, idVendor=0a5c, idProduct=2711, bcdDevice= 0.00
[  +0.000019] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000015] usb 1-1.1: Product: BCM2711 Boot
[  +0.000013] usb 1-1.1: Manufacturer: Broadcom
[  +0.000012] usb 1-1.1: SerialNumber: 74b0ffaa
[  +0.234984] usb 1-1.2: new high-speed USB device number 53 using xhci_hcd
[  +0.100504] usb 1-1.2: config index 0 descriptor too short (expected 55, got 32)
[  +0.000508] usb 1-1.2: New USB device found, idVendor=0a5c, idProduct=2711, bcdDevice= 0.00
[  +0.000023] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000017] usb 1-1.2: Product: BCM2711 Boot
[  +0.000015] usb 1-1.2: Manufacturer: Broadcom
[  +0.000014] usb 1-1.2: SerialNumber: b91398e2
[  +1.869797] usb 1-1.1: USB disconnect, device number 52
[  +3.293209] usb 1-1.1: new high-speed USB device number 54 using xhci_hcd
[  +0.100821] usb 1-1.1: New USB device found, idVendor=0a5c, idProduct=2711, bcdDevice= 0.00
[  +0.000039] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=4
[  +0.000018] usb 1-1.1: Product: BCM2711 Boot
[  +0.000015] usb 1-1.1: Manufacturer: Broadcom
[  +0.000013] usb 1-1.1: SerialNumber: 74b0ffaa
[  +0.003556] usb 1-1.2: USB disconnect, device number 53
[  +2.235330] usb 1-1.1: USB disconnect, device number 54
[  +0.228151] usb 1-1.1: new high-speed USB device number 55 using xhci_hcd
[  +1.095995] usb 1-1.1: new full-speed USB device number 56 using xhci_hcd
[  +0.101934] usb 1-1.1: not running at top speed; connect to a high speed hub
[  +0.008006] usb 1-1.1: New USB device found, idVendor=0a5c, idProduct=2764, bcdDevice= 0.00
[  +0.000029] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=1
[  +0.000015] usb 1-1.1: Product: BCM2711 Boot
[  +0.000013] usb 1-1.1: Manufacturer: Broadcom
[  +0.000012] usb 1-1.1: SerialNumber: Broadcom
[  +0.230007] usb 1-1.2: new high-speed USB device number 57 using xhci_hcd
[  +0.100847] usb 1-1.2: New USB device found, idVendor=0a5c, idProduct=2711, bcdDevice= 0.00
[  +0.000031] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=4
[  +0.000017] usb 1-1.2: Product: BCM2711 Boot
[  +0.000014] usb 1-1.2: Manufacturer: Broadcom
[  +0.000014] usb 1-1.2: SerialNumber: b91398e2
[Apr14 15:29] usb 1-1.2: USB disconnect, device number 57
[  +0.222338] usb 1-1.2: new high-speed USB device number 58 using xhci_hcd
[  +1.024004] usb 1-1.2: new full-speed USB device number 59 using xhci_hcd
[  +0.101964] usb 1-1.2: not running at top speed; connect to a high speed hub
[  +0.006007] usb 1-1.2: New USB device found, idVendor=0a5c, idProduct=2764, bcdDevice= 0.00
[  +0.000026] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=1
[  +0.000015] usb 1-1.2: Product: BCM2711 Boot
[  +0.000013] usb 1-1.2: Manufacturer: Broadcom
[  +0.000012] usb 1-1.2: SerialNumber: Broadcom
[  +2.485798] usb 1-1.1: USB disconnect, device number 56
[  +0.366136] usb 1-1.1: new high-speed USB device number 60 using xhci_hcd
[  +5.148292] usb 1-1.1: device descriptor read/64, error -110
[ +15.616052] usb 1-1.1: device descriptor read/64, error -110
[  +0.187802] usb 1-1.1: new high-speed USB device number 61 using xhci_hcd

hobbit23, 2023-04-14:
> > even the "fresh" CM4 modules can not be connected by the server
> 
> Make sure the boot switch of fresh modules is set to "eMMC boot" "ON" so that they can network boot.

now this worked! 
fresh CM4 and boot switch to ON and external USB power supply and only ETH connection to the server

hobbit23, 2023-04-19:
Problem: 
I connected four CM4 already programmed via USB Hub (powered) to the USB port of the server. Also they are connected to the switch via ETH. They are booting up and trying to get flashed but the process is not finishing after 4 hours now....what is wrong here?

maxnet, 2023-04-19:
Got a photo of the HDMI output of those CM4s?

hobbit23, 2023-04-19:
[image]
[image]
[image]

maxnet, 2023-04-19:
Meant the CM4 boards being provisioned, not the Pi 4 server.

maxnet, 2023-04-20:
>O do not have access to them because they are not ready flashed

A Linux operating system is network booted on them during provisioning.
So they do should display SOMETHING when you connect HDMI to them....

hobbit23, 2023-04-21:
sorry, i don't understand what you mean.

I am using a windows computer to add the file to the USB stick.

At a different system, where I worked before, i used linux and can flash the CM4 with the bmap tool which is running must faster and skippes the free spaces.

maxnet, 2023-04-21:
>sorry, i don't understand what you mean.

I asked you to connect a HDMI display to the CM4 boards.
With that I mean your Sourcekit boards.

You show me 3 pictures that are of a HDMI display that is clearly connected to the Pi 4 that is playing server instead. Messages like "Product: BCM2711 boot" will only appear on the server, definitely not on the client.
That is not what I need.
If you want support, please follow instructions to the letter. And connect that HDMI display to one of the CM4/Sourcekit boards, and see what it shows on screen.
If the boards manage to network boot, and your start post suggest they do (otherwise it wouldn't show "starting to write image"), there should definitely be something on the screen.

>At a different system, where I worked before, i used linux and can flash the CM4 with the bmap tool which is running 
>must faster and skippes the free spaces.

If you have a lot of free space in your images, then you are generating images the wrong way.
Make sure you use tools like pi-gen if what you have is based on Pi OS.
But that is a different discussion. And not relevant to your current issue.

hobbit23, 2023-04-21:
I am using a yocto image so no Pi OS. we can skip this for now

hobbit23, 2023-04-24:
"I asked you to connect a HDMI display to the CM4 boards."

nothing appears on the HDMI screen. I can understand this because the eMMC is not flashed

maxnet, 2023-04-24:
>I can understand this because the eMMC is not flashed

I think you are misunderstanding how cmprovision works.

We do NOT use the normal rpiboot payload that lets the cm4 act as a USB mass storage storage device to flash it.

We start a small Linux operating system first (made with buildroot, similar to Yocto) instead.
That small operating system contacts the cmprovision server for further instructions, and downloads and flashes the image from the server.
If you see messages in the webinterface of the server like "starting to write image" it means the small operating system has had contact with the server, so must have started.
And the small operating system certainly outputs to HDMI.
If you do not see anything on HDMI while you do see messages in the webjnterface, either your boards are broken, or there is something wrong with your HDMI display (which is unlikely because your HDMI display does seem to work when connected to the server)

hobbit23, 2023-04-24:
thanks for clarification.
I already checked with some other boards I am using here. I can flash them at my laptop with a separate tool but after connection to the server and wait a few seconds, the HDMI still is no video visible.

When is the small operating system programmed to the module? How long does it take and how can this be indicated from outside?

maxnet, 2023-04-24:
>When is the small operating system programmed to the module?

The small operating system gets booted to RAM (using an initramfs construct), it does not get written to flash.
It is stored inside scriptexecute.img that you can see getting downloaded by tftp in the dnsmasq log if network booted (or the rpiboot log if USB booted)

The moment you see messages like "Provisioning started" in the web interface, the small OS is already running.
The web interface writes that to log if it receives a HTTP request from the small OS running on the client asking for instructions.

hobbit23, 2023-04-24:
thanks for the explanation. 
The boards are powered like in any other flashing process so they have enough power I expect.
I will try to use an decompressed image...

maxnet, 2023-04-24:
>The boards are powered like in any other flashing process so they have enough power I expect.

Recall that rpiboot's mass storage device stuff runs on the Videocore GPU.
Our utility operating system runs on the ARM CPU.
Different power footprint, and Pi 4 can only deliver 1.2 amperre for all USB ports combined.

hobbit23, 2023-04-24:
"We are not "any other flashing process""
-> I mean I can also flash the CM4 with another way of flashing method where the CM4 needs power as well. I don't use the raspi USB for power the CM4

hobbit23, 2023-04-26:
Hello,
I want to come back to the small operating system which is neccessary on the CM4 to get the image flashed by the server. You mentioned, that the CMs seems to have the small operating system installed because they show up in the log.

One step back: I expected that the server is running, and as soon as I connect a device via ETH to the ETH-switch the device is flashed. Even, when I connect more devices to the same ETH-swtich afterwards, the progress of the first one will continue and the second device will be flashed as well. 
Can you acknowledge this expectation?

Can we find out together why my system is not running like it should? Is there any chance to get support by you on a different way? Is it possible to come to your location to verify the setup? Can we do a remote session or maybe you can tell me to share more details about my system?
Thanks for your help!

ghollingworth, 2023-05-05:
As mentioned elsewhere use a separate device to wipe the CM4, cmprovision is only developed for provisioning of new devices.

closed 2023-05-05 by ghollingworth