raspberrypi/cmprovision #23: How to use the scripts

hobbit23, 2023-04-26:
Hello,
i want to use the scripts to get connected to an online server after flashing. Is this possible and is there any example i can try?
Can the script start AFTER flashing the CM4?

hobbit23, 2023-05-05:
-	Is it possible to start a script after flashing process?
-	Is it possible to a enumeration as a serial number to the devices?
-	Can I add a certificate during flashing process?

maxnet, 2023-05-05:
>Is it possible to start a script after flashing process?

When adding a script in the web interface you can choose between "pre" and "post" installation.
"pre" means before flashing.
"post" means after flashing.

>Is it possible to a enumeration as a serial number to the devices?

Not functionality we provide by default.
cmprovision includes source code though, and you can hire any random PHP developer of your choice, to add such things to your requirements.

>Can I add a certificate during flashing process?

I am afraid I do not know what type of certificate you are interested in.
We do not support programming secure boot keys/certificates, if that is what you mean.
Do can generate client-side certificates typically used to communicate with an IoT/cloud service.

If you need an example of a script that does X, I suggest you ask on the forum though.
This are not cmprovision bugs for which the github issue tracker is meant.
Keep in mind that for every issue you create here, everyone that is subscribed to this repository gets an e-mail notification.