raspberrypi/cmprovision #24: Wrong Disk space available

giampiero7, 2023-05-26:
Hi,
When adding a new image the popup shows that I have 0.1 GiB available, while in fact I have more than 20 GiB.
How is the available space calculated?
And where are the images stored in the filesystem?

Thanks!

maxnet, 2023-05-26:
Uses: https://www.php.net/manual/en/function.disk-free-space.php

>And where are the images stored in the filesystem?

Somewhere under /var/lib/cmprovision/storage

You do are using the 64-bit edition of RPI OS?
Any calculations done by PHP on large numbers will NOT work properly on 32-bit operating systems.

giampiero7, 2023-05-29:
Hi,
Yes, I'm using the 64-bit version.
On the installation I was using, I had the path `/var/log` configured as `tmpfs` in `/etc/fstab`.
To attempt to find the issue I started from a newly flashed image, long story short: if I remove the dir `/var/log/nginx` and reboot, cmprovision stops working as nginx.service fails to start:

```
$ systemctl status --no-pager -l nginx.service
● nginx.service - A high performance web server and a reverse proxy server
     Loaded: loaded (/lib/systemd/system/nginx.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Mon 2023-05-29 16:26:11 BST; 11min ago
       Docs: man:nginx(8)
    Process: 534 ExecStartPre=/usr/sbin/nginx -t -q -g daemon on; master_process on; (code=exited, status=1/FAILURE)
        CPU: 98ms

May 29 16:26:11 flasher-tool-001 systemd[1]: Starting A high performance web server and a reverse proxy server...
May 29 16:26:11 flasher-tool-001 nginx[534]: nginx: [alert] could not open error log file: open() "/var/log/nginx/error.log" failed (2: No such file or directory)
May 29 16:26:11 flasher-tool-001 nginx[534]: 2023/05/29 16:26:11 [emerg] 534#534: open() "/var/log/nginx/access.log" failed (2: No such file or directory)
May 29 16:26:11 flasher-tool-001 nginx[534]: nginx: configuration file /etc/nginx/nginx.conf test failed
May 29 16:26:11 flasher-tool-001 systemd[1]: nginx.service: Control process exited, code=exited, status=1/FAILURE
May 29 16:26:11 flasher-tool-001 systemd[1]: nginx.service: Failed with result 'exit-code'.
May 29 16:26:11 flasher-tool-001 systemd[1]: Failed to start A high performance web server and a reverse proxy server.
```

It looks like it doesn't create the dir `/var/log/nginx` if it does not exist.

This might or might not be related to the original issue, which I didn't manage to reproduce.

giampiero7, 2023-05-29:
Update: the original issue was due to `/tmp` being configured as `tmpfs` with limited space in `/etc/fstab`.
So, with the following in `/etc/fstab` you create both issues in one go:

```
proc            /proc           proc    defaults          0       0
PARTUUID=aaaf927c-01  /boot           vfat    defaults          0       2
PARTUUID=aaaf927c-02  /               ext4    defaults,noatime  0       1
tmpfs           /tmp            tmpfs   noatime,size=64m      0       0
tmpfs           /var/log        tmpfs   noatime,size=16m      0       0
```