raspberrypi/cmprovision #26: bcm2711-rpi-cm4s.dtb is missing for CM4S.  CM4S provisioning fails to boot scriptexecute firmware.

JinShil, 2023-07-17:
The bcm2711-rpi-cm4s.dtb file is missing from the scriptexecute firmware.  Without it, the scriptexecute firmware fails to boot.  With it, the scriptexecute firmware boots, but does not start the provisioning process, but it appears to be a step in the right direction.

maxnet, 2023-07-17:
>With it, the scriptexecute firmware boots, but does not >start the provisioning process.

I am afraid I do not have a cm4s at the moment, so cannot test.

But looking at the console output in your forum post, it has the wrong cmdline for using USB to communicate with server, so it is waiting for eth0 instead.
So it seems your [cm4s] config.txt section is not being applied.

JinShil, 2023-07-17:
Thank you.  As stated later in the forum thread, adding the following the config.txt file allowed me to provision the CM4S over a USB cable.
```
[board-type=0x15]
dtoverlay=dwc2,dr_mode=peripheral
cmdline=cmdline.txt.ipv6ll
```
I'm assuming the scriptexecute firmware in the latest cmprovision software is old, so it did not recognize the `[cm4s]` filter.

JamesH65, 2023-07-17:
@pelwell FYI

pelwell, 2023-07-17:
I'm not sure what I am supposed to take from this issue and the associated forum thread, other than it looks like at least some CM4s boards have been manufactured with an EEPROM image that pre-dates the addition of support for the "[cm4s]" section. As @JinShil has shown, `[board-type=0x15]` is a suitable alternative.

The absence of `bcm2711-rpi-cm4s.dtb` from this repo is one for @maxnet.

JinShil, 2023-07-17:
I think its the scriptexecute firmware in the latest version of cmprovision that does not recognize the `[cm4s]` filter.

JinShil, 2023-09-08:
Can we please get an update to the scriptexecute firmware in this repository?