raspberrypi/cmprovision #27: seeed device 'waiting for BCM2835/6/7/2711'

yspud, 2023-09-01:
I am an admitted newb. trying to set up an easy way to image 1000+ cm4 units.  I can manually flash each using rpiboot and pi imager software but would love to automate this with cmprovision.  The logs just get to the 'second stage boot' and then hang on the message waiting for BCM2835/6/7/2711 and thats it.   Not sure what else to do.  Any help much appreciated.  thank you so much .

maxnet, 2023-09-01:
Network booting CM4s with empy eMMC does work properly?

yspud, 2023-09-01:
the eMMC comes shipped with an OS and i couldnt get network boot to work.  its attached to a screen but i think there is a mini-hdmi out - would connecting that provide more boot data ?  the integrated 5 inch screen just stays black so i have very little indication of what the device is doing other than a couple of status lights (which go off when in the 'waiting for bcm' stage)  

this is the device fyi:  https://www.seeedstudio.com/ReTerminal-with-CM4-p-4904.html

only has a single usb-c for power/data so i tried a rasPi 4 direct usb ports and even got a 60W external usb-c hub just in case it was a power issue but same issue. ..  thank you so much for the reply.