raspberrypi/cmprovision #28: public/uploads directory missing when built from source

ysica, 2023-09-19:
I built cmprovision from source using dpkg-buildpackage. I then compared my local deb against the 1.6.3 release and noticed that `/var/lib/cmprovision/public/uploads` was missing from my local deb. This causes the /images endpoint to return a 500 error ([Images.php:21](https://github.com/raspberrypi/cmprovision/blob/main/app/Http/Livewire/Images.php#L21)).

I suspect this is because the `uploads` directory is not included in the Git source. I solved the problem by creating a debian/dirs file and adding the directory to it, but the [documentation](https://www.debian.org/doc/manuals/maint-guide/dother.en.html#dirs) suggests this is generally not desirable.

Should I be building the source differently, or is this a bug that a PR should be submitted for?

barneyman, 2023-11-23:
My build script does this effectively ...
```
mkdir public/uploads
dpkg-buildpackage -us -uc --build=all
```

ysica, 2023-11-23:
Thanks barney, I solved the issue locally in a similar way. I created the `debian/dirs` file ([documentation](https://www.debian.org/doc/manuals/maint-guide/dother.en.html#dirs)) with the contents:

`var/lib/cmprovision/public/uploads`