raspberrypi/cmprovision #29: NVMe Support

Panxatony, 2023-09-21:
If I recognize correctly, the network boot image does not support NVMe. There is currently an exciting project https://computeblade.com/ and support for NVMe alongside eMMC would be very helpful.

timg236, 2023-09-21:
Off topic; the CM4 rpiboot mass storage gadget supports NVMe. Slightly more manual process though.

maxnet, 2023-09-21:
>If I recognize correctly, the network boot image does not support NVMe. 

Yes, cmprovision is older than the Pi had NVMe support.

Should work if you toss CONFIG_BLK_DEV_NVME=y in https://github.com/raspberrypi/scriptexecutor/blob/master/scriptexecute/configs/scriptexecute_cm4_defconfig and recompile.
(Or steal the kernel from Imager embedded).

Panxatony, 2023-09-23:
> > If I recognize correctly, the network boot image does not support NVMe.
> 
> Yes, cmprovision is older than the Pi had NVMe support.
> 
> Should work if you toss CONFIG_BLK_DEV_NVME=y in https://github.com/raspberrypi/scriptexecutor/blob/master/scriptexecute/board/kernelconfig-cm4.armv7 and recompile the Linux kernel. (Or steal the kernel from Imager embedded and toss it in /var/lib/cmprovision/scriptexecute ).

That was the hint I was missing. Thank you very much, it worked.

Panxatony, 2023-09-23:
That was the hint I was missing. Thank you very much, it worked.

closed 2023-09-23 by Panxatony