raspberrypi/cmprovision #32: Provisioning fails - Error during dd. Return code 1.

Dude90, 2023-11-02:
Hi,

we have been using the CM-Provision for quite a while now and it is a great tool. Thank you for that!

We never had any problems provisioning our CM4 modules until a few weeks ago. Since the tool version we used still had the MAC-filter, we did ran into problems with a new batch of CM4. Hence, I updated our provision station to the newest version 1.6.3 (without the MAC-filter) which seemed to fixed the problem. After the update we were able to flash ~15 CM4 modules without any problems. However since about a week nearly all attempts to flash new CM4 boards are suddenly resulting in the following error:

[image]

Here a corresponding detail-page of a failing CM4:

[image]

Before we never encountered this error. In the last week 15 out of 20 CM4 boards showed it.

I already had a look through the existing issues. While issue #15 seems to talk about a similar error, the proposed solution does not fit to our problem since our image only has an uncompressed size of 4.3GB while the CM4 modules provide 32GB of storage.

We would really appreciate, if somebody could help us.

Thank you!

tdewey-rpi, 2023-11-02:
Tagging @maxnet, can you please take a look at this? Curious that it's only writing out 4K records each time.

maxnet, 2023-11-02:
> I already had a look through the existing issues. While issue https://github.com/raspberrypi/cmprovision/issues/15
> seems to talk about a similar error, the proposed solution does not fit to our [...]

What is the exact uncompressed size in bytes?

We are using 1 MiB output block size.
So 4096+1 records, would be 4096 MiB + a tiny amount more that is not a whole megabyte.
You are using an image that is not dividable by whole megabytes?

What does pre-install script output say?
Any eMMC customization (such as pSLC mode)?

Dude90, 2023-11-02:
The exact uncompressed size is: 4.294.967.296 Bytes
The image is dividable in whole megabytes (1024 Byte), at least in my opinion.
I can try it with a normal RPI OS Image, but we have not changed our image (or something else). Before the problem appeared, the flashing of that exact image (even with the same project) worked perfectly fine. The problem also them to occurs randomly. Some CM4 board can be flashed without problems, other not. If a failed board is recovered using an additional USB connection to the provision station, the flashing then often works.

The pre-install script currently only turns on a LED on the CM4 carrier we are using. This works fine.
We do not have any eMMC customization. We are using regular CM4 boards, straight from the distributor.

If I can to anything to help debug this problem, please let me know.

maxnet, 2023-11-02:
If you attach a HDMI monitor to the board being provisioned, does it show anything out of the ordinary?
E.g. curl having to retry the download (not sure if that works correctly in all cases)?
Or kernel messages indicating an IO error?

Dude90, 2023-11-03:
@maxnet I checked the HDMI output during a failing flashing attempt. Everything looks normal to me but in the end it says:

`
gzip: crc error
Writing image failed.
`

Below you can find a picture of the output.
We are using gzip (gz) compression for our image.

Presented by only this information I would assume a corrupted image file. However, the same image file worked perfectly fine and even in the last week a few boards could be flashed using this image without a problem.

[image]

maxnet, 2023-11-03:
Suggest you test with our stock images first.
If issue persist you may also want to double check if the boards affected run stable under load.
E.g. install RPI OS lite, and try:

```
sudo apt update
sudo apt install memtester
memtester 1500M
```

Let it run for a while and see if it detects any errors.
Make sure you test with same power supplies as you use during provisioning.

Dude90, 2023-11-03:
@tdewey-rpi  Unfortunately I can currently not record a video of my workflow, but I can give you the following description with pictures:

**Flashing:**

Setup:
[image]
We are using a RPI 4 B as provision station. The RPI is connected to a switch and has a second USB ETH interface for connection to our network. In this example I am using the CM4IO board as a carrier for the CM4 board that should be flashed.

HDMI-Output:
[image]

Webpage-Log:
[image]

**Recovery:**

Setup:
[image]
For recovery I have set the jumper to disable eMMC boot and I additionally connected the CM4 to the provision station using USB.

HDMI-Output:
[image]

Webpage-Output:
[image]

Here are the logs corresponding to the flashing attempts shown above:
[log_dnsmasq.txt](https://github.com/raspberrypi/cmprovision/files/13249917/log_dnsmasq.txt)
[log_rpiboot.txt](https://github.com/raspberrypi/cmprovision/files/13249916/log_rpiboot.txt)
[log_dmesg.txt](https://github.com/raspberrypi/cmprovision/files/13249919/log_dmesg.txt)
[log_syslog.txt](https://github.com/raspberrypi/cmprovision/files/13249921/log_syslog.txt)

After the flashing and the recovery failed I used the USB-mass-storage mode to attach the CM4 to my computer and directly transfered an RPI OS Lite Image onto the board. This worked perfectly fine and I could normally start the CM4 afterwards and access the system.
@maxnet On this system I also performed a short run of the memtester (only two cycles). This produced no error.

Next I will exchange the image in the provision station with the official RPI OS Lite image and check if the provision process will work afterwards.

Dude90, 2023-11-03:
Ok, after replacing our image with the official RPI OS Lite image the provisioning worked again, even for a board that failed before.

Just to give it a try, I freshly reuploaded our image and set the newly uploaded image _active_. To my surprise the provisioning also worked fine with this image. If I switch back to the old upload of our image, the provisioning fails again, even though the image files are exactly the same (even the computed hash is exactly the same - see below).

[image]

In my opinion, the only explanation for this behavior is that the old image file somehow got damaged after uploading (and using it for a while). Is this possible? Can you tell me where the images are stored on the provisioning device? Then I can recompute the hash of both the old and the newly uploaded image and diff them.

maxnet, 2023-11-03:
Not near a cmprovision installation but recall /var/lib/cmprovision/storage/app/public
Search for the filename mentioned in your photo xAth ... .gz

And yes, SD card storage is fragile.
It cannot do small writes. If you modify anything on SD card it will have to read a larger block of data, change the bytes you want to change, and write the large block of data back to storage.

Dude90, 2023-11-03:
Thank you! With your hint I was able to find the images. The exact location of the image storage is _/var/lib/cmprovision/public/uploads_.

The hashes are confirming the suspicion. They differ between the old image and newly uploaded image. While the hash of the newly uploaded image equals the hash that I have computed on my computer, the hash of the old image does not. Hence, the old image has indeed somehow been damaged.
But I am still wondering why a few boards could nevertheless be successfully flashed with the old image. That seems strange... Do you have any explanation for that?

maxnet, 2023-11-04:
>But I am still wondering why a few boards could nevertheless be successfully flashed with the old image. That seems 
>strange... Do you have any explanation for that?

If that happened in quick succession (one board fails, one board works, after each other, without you rebooting the host computer) there is also a possibility there was a memory error on your host Pi.
As I would expect your 257 MB compressed image to be still in memory (Linux' buffer cache) and not re-read from storage in that case.
Unless you are using a model with very little memory.
However then the problem would likely to be gone after a reboot though.

Problems with your SD card are more likely.
Cheap SD storage sticks 3 bits in a cell.
When reading from it, it tests how well current flows through the memory cell, and has to distinguish between 8 possible voltage levels.
May be very close to the threshold voltage of a different level, resulting in different results on different reads.

(Some flash storage can be programmed to stick less bits in a cell, giving better reliability at a cost of permanently losing capacity.
The eMMC on the CM4 boards does supports that. It is what the option "Format eMMC as pSLC (one time settable only)" in cmprovision does.
But very few normal SD cards do, so will not help on your host system.
Do note you can also install cmprovision on a normal x86 computer with normal disks. It is not mandatory to use a Pi).

closed 2023-11-06 by Dude90