raspberrypi/cmprovision #36: Issue with CM4 Programming via Ethernet

francescoriga92, 2024-01-24:
Hello everyone,

I'm encountering a problem while trying to program my CM4 (with eMMC) board via Ethernet. When I connect both Ethernet and USB cables, everything works fine. However, when I try programming solely through Ethernet, it doesn't seem to be working (with the cm4 card formatted). I suspect it might be a formatting issue causing the master to not recognize the board and initiate programming via Ethernet.

Has anyone else experienced a similar problem? I would greatly appreciate any insights or solutions you might have. Thank you in advance for your help!

Best regards,
[image]

JamesH65, 2024-01-24:
To program via Ethernet the device needs to be "factory fresh", i.e. as if it has just come from the factory or has been reset to factory settings.  I suspect you have something already programmed in?

francescoriga92, 2024-01-24:
Hi JamesH65, Yes, it's true that the CM4 board was previously programmed. However, I attempted to reformat it using the filesystem FAT32 and tried to restore it to factory conditions. What is the correct formatting method to bring it back to factory settings?

timg236, 2024-01-24:
The boot-order says 0xf641 which indicates that this is not the factory boot order.
You can reset the EEPROM via rpiboot
https://github.com/raspberrypi/usbboot/blob/master/recovery/README.md

maxnet, 2024-01-24:
>You can reset the EEPROM via rpiboot

The TS already has cmprovision setup, so it may be easier if you explain what boot order should be put in the "EEPROM settings" field there...

timg236, 2024-01-24:
Well presumably you'd include 0x2 if you wanted network boot to work e.g. 0xf2641

SD -> USB -> NVME -> Network

francescoriga92, 2024-01-26:
Thanks everyone, I did it! I reset the eeprom via rpiboot and everythings works now.

maxnet, 2024-01-26:
>Do you know how many devices I can flash at a time? 

512

>All devices after 5 give me the error shown in the photo

Can you connect HDMI to one of the boards being provisioned, and take a photo of any error there?