raspberrypi/cmprovision #37: CM3+ Flashing / Provisioning Does Not Seem to Work

stevenm3656, 2024-02-01:
Hello, hoping to get some help.

Perhaps I just haven't left it long enough, but not sure it should take more than a few hours.

Setup:

Provisioner:
Pi 4b 8GBram, 64GB micro SD with latest 64bit Raspberry Pi OS Lite (as of 31/01/2024 - yesterday)

Web interface working (over WiFi), images uploaded and projects set up.

Tried to provision / flash a CM3+ 32GB:

-  connected the Slave USB on the official IO Board to the Pi 4
- powered on CM3+
- from the logs it looks like this is detected as I can see the bootloader messages you would normally see from running rpiboot on windows etc

Pi doesn't appear to be flashed with image.

Balena etcher would take around 16 minutes to flash this image, what should I expect from this?
Is there any steps for CM3+ not listed on this repo? Can't find much online.
Is there any other logs that would show anything after the bootloader stuff?

Many thanks,

Steven

maxnet, 2024-02-02:
If it is a recent RPI OS version you may need to swap out networkmanager for dhcpcd (via sudo raspi-config).

Other than that connect HDMI to the boards you are provisioning and check screen for errors.

stevenm3656, 2024-02-02:
Thanks,

No option for selecting between networkmanager or dhcpcd on the Pi 4.

Connected HDMI to CM3+, got:

... random: fast init done
curl: (7) Failed to connect to ... port 80: Connection timed out

Welcome to Buildroot
scriptexecute login:

maxnet, 2024-02-02:
>No option for selecting between networkmanager or dhcpcd on the Pi 4.

May need to install an older RPI OS version (Bullseye) for that.

Or figure out how you can DISABLE IPv6 privacy extensions on NetworkManager. (Do must be set globally, because every time a new module is being provisioned the usb network adapter will change. So network manager per connection profiles are not going to do.)

The modules being provisioned expect that they are able to reach the server on an predictable IPv6 link local address, that can be calculated based on the MAC address of the network adapter (EUI-64 method).
And if Network Manager is randomizing the IPv6 address for privacy reasons instead, the module has no clue what it needs to connect to.

stevenm3656, 2024-02-05:
Thanks Max,

You pointed me in the right direction, all sorted now and the provisioner is excellent, thanks a lot to all who worked on it.

I'll detail the fix for anyone else facing the same issue.

I was able to stay on the latest version and follow instructions from [here](https://forums.raspberrypi.com/viewtopic.php?t=360586), I'll also add the steps below for completeness.

I then had to edit `/etc/dhcpcd.conf` as per the instructions on the [Provisioning from an alternate OS section.](https://github.com/raspberrypi/cmprovision#provisioning-from-an-alternate-os), I'll include this in the steps below.

1. `sudo systemctl disable NetworkManager`
2. `sudo systemctl stop NetworkManager`
3. `sudo apt-get update`
4. `sudo apt-get install -y dhcpcd5`
5. `sudo cp /usr/share/dhcpcd/hooks/10-wpa_supplicant /lib/dhcpcd/dhcpcd-hooks/10-wpa_supplicant`
6. `sudo nano /etc/wpa_supplicant/wpa_supplicant.conf`

Key in your wifi credential as usual

```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=GB

network={
  ssid="<Enterprise-SSID>"
  key_mgmt=WPA-EAP
  eap=PEAP
  identity="<username>"
  password=hash:<hashed-password>
}
```

7. `sudo nano /etc/dhcpcd.conf`

At the bottom of the file un-comment `#slaac hwaddr` and comment `slaac private` so you have:

```
# Generate SLAAC address using the Hardware Address of the interface
slaac hwaddr
# OR generate Stable Private IPv6 Addresses based from the DUID
#slaac private
```

9. `sudo reboot`

closed 2024-02-05 by stevenm3656