raspberrypi/cmprovision #39: What is available to scripts?

Siecje, 2024-02-05:
Specifically I'm wondering if rpi-otp-private-key is available in a post install script.

ghollingworth, 2024-02-05:
The thing executing code on the target is the scriptexecute buildroot, you can see a compiled version of that in the scriptexecute directory.  It comes from https://github.com/raspberrypi/scriptexecutor

You should be able to rebuild this and add anything you need to it...

Otherwise, if the rpi-otp-private-key script only calls stuff that's already inside the scriptexecutor build then you could just copy-paste it into the postinstall script...

maxnet, 2024-02-05:
>What is available to scripts?

By default:

```
cmprovision/scriptexecute$ xz -dc scriptexecute.img | cpio -ivt
drwxr-xr-x  17 root     root            0 Nov 10  2022 .
drwxr-xr-x   4 root     root            0 Nov 10  2022 dev
lrwxrwxrwx   1 root     root           17 Jan 12  2020 dev/stderr -> ../proc/self/fd/2
lrwxrwxrwx   1 root     root           17 Jan 12  2020 dev/stdout -> ../proc/self/fd/1
drwxr-xr-x   2 root     root            0 Jan 12  2020 dev/pts
lrwxrwxrwx   1 root     root           10 Jan 12  2020 dev/log -> ../tmp/log
lrwxrwxrwx   1 root     root           17 Jan 12  2020 dev/stdin -> ../proc/self/fd/0
drwxr-xr-x   2 root     root            0 Jan 12  2020 dev/shm
crw--w--w-   1 root     root       5,   1 Nov 10  2022 dev/console
lrwxrwxrwx   1 root     root           15 Jan 12  2020 dev/fd -> ../proc/self/fd
drwxr-xr-x   2 root     root            0 Jan 12  2020 opt
drwxr-xr-x   4 root     root            0 Nov 10  2022 var
lrwxrwxrwx   1 root     root            6 Jan 12  2020 var/cache -> ../tmp
lrwxrwxrwx   1 root     root            6 Jan 12  2020 var/spool -> ../tmp
lrwxrwxrwx   1 root     root            6 Jan 12  2020 var/log -> ../tmp
drwxr-xr-x   2 root     root            0 Jan 12  2020 var/lib
lrwxrwxrwx   1 root     root            9 Jan 12  2020 var/lib/misc -> ../../tmp
lrwxrwxrwx   1 root     root            6 Jan 12  2020 var/run -> ../run
lrwxrwxrwx   1 root     root            6 Jan 12  2020 var/tmp -> ../tmp
drwxr-xr-x   2 www-data www-data        0 Nov 10  2022 var/www
lrwxrwxrwx   1 root     root            6 Jan 12  2020 var/lock -> ../tmp
drwxr-xr-x   2 root     root            0 Nov 10  2022 bin
lrwxrwxrwx   1 root     root            7 Apr  7  2021 bin/mv -> busybox
lrwxrwxrwx   1 root     root            7 Apr  7  2021 bin/su -> busybox
lrwxrwxrwx   1 root     root            7 Apr  7  2021 bin/false -> busybox
lrwxrwxrwx   1 root     root            7 Apr  7  2021 bin/stty -> busybox
-rwsr-xr-x   1 root     root       772600 Nov 10  2022 bin/busybox
lrwxrwxrwx   1 root     root            7 Apr  7  2021 bin/link -> busybox
lrwxrwxrwx   1 root     root            7 Apr  7  2021 bin/setserial -> busybox
lrwxrwxrwx   1 root     root            7 Apr  7  2021 bin/printenv -> busybox
lrwxrwxrwx   1 root     root            7 Apr  7  2021 bin/rm -> busybox
lrwxrwxrwx   1 root     root            7 Apr  7  2021 bin/kill -> busybox
-rwxr-xr-x   1 root     root         9408 Nov 10  2022 bin/lsattr
lrwxrwxrwx   1 root     root            7 Apr  7  2021 bin/date -> busybox
-rwxr-xr-x   1 root     root         1342 Apr  7  2021 bin/compile_et
lrwxrwxrwx   1 root     root            7 Apr  7  2021 bin/hostname -> busybox
lrwxrwxrwx   1 root     root            7 Apr  7  2021 bin/gunzip -> busybox
lrwxrwxrwx   1 root     root            7 Apr  7  2021 bin/netstat -> busybox
lrwxrwxrwx   1 root     root            7 Apr  7  2021 bin/base64 -> busybox
-rwxr-xr-x   1 root     root        50952 Nov 10  2022 bin/dmesg
lrwxrwxrwx   1 root     root            7 Apr  7  2021 bin/sync -> busybox
lrwxrwxrwx   1 root     root            7 Apr  7  2021 bin/setarch -> busybox
-rwxr-xr-x   1 root     root        47744 Nov 10  2022 bin/findmnt
lrwxrwxrwx   1 root     root            7 Apr  7  2021 bin/pipe_progress -> busybox
lrwxrwxrwx   1 root     root            7 Apr  7  2021 bin/ls -> busybox
lrwxrwxrwx   1 root     root            7 Apr  7  2021 bin/mknod -> busybox
lrwxrwxrwx   1 root     root            7 Apr  7  2021 bin/nice -> busybox
-rwxr-xr-x   1 root     root         1102 Apr  7  2021 bin/mk_cmds
lrwxrwxrwx   1 root     root            7 Apr  7  2021 bin/pwd -> busybox
lrwxrwxrwx   1 root     root            7 Apr  7  2021 bin/usleep -> busybox
lrwxrwxrwx   1 root     root            7 Apr  7  2021 bin/setpriv -> busybox
lrwxrwxrwx   1 root     root            7 Apr  7  2021 bin/more -> busybox
lrwxrwxrwx   1 root     root            7 Apr  7  2021 bin/mount -> busybox
lrwxrwxrwx   1 root     root            7 Apr  7  2021 bin/arch -> busybox
lrwxrwxrwx   1 root     root            7 Apr  7  2021 bin/mt -> busybox
lrwxrwxrwx   1 root     root            7 Apr  7  2021 bin/touch -> busybox
-rwxr-xr-x   1 root     root         9408 Nov 10  2022 bin/chattr
lrwxrwxrwx   1 root     root            7 Apr  7  2021 bin/grep -> busybox
lrwxrwxrwx   1 root     root            7 Apr  7  2021 bin/getopt -> busybox
lrwxrwxrwx   1 root     root            7 Apr  7  2021 bin/mkdir -> busybox
lrwxrwxrwx   1 root     root            7 Apr  7  2021 bin/resume -> busybox
lrwxrwxrwx   1 root     root            7 Apr  7  2021 bin/cat -> busybox
lrwxrwxrwx   1 root     root            7 Apr  7  2021 bin/rmdir -> busybox
lrwxrwxrwx   1 root     root            7 Apr  7  2021 bin/dumpkmap -> busybox
[... 517 lines cut ...]
-rw-r--r--   1 root     root          340 Jan 12  2020 etc/passwd
-rw-r--r--   1 root     root          334 Jan 12  2020 etc/fstab
-rw-r--r--   1 root     root          512 Apr  3  2021 etc/random-seed
drwxr-xr-x   6 root     root            0 Apr  7  2021 etc/network
-rw-r--r--   1 root     root           77 Apr  7  2021 etc/network/interfaces
drwxr-xr-x   2 root     root            0 Jan 12  2020 etc/network/if-down.d
drwxr-xr-x   2 root     root            0 Nov 10  2022 etc/network/if-pre-up.d
-rwxr-xr-x   1 root     root          576 Jan 12  2020 etc/network/if-pre-up.d/wait_iface
drwxr-xr-x   2 root     root            0 Jan 12  2020 etc/network/if-post-down.d
[... cut ...]
-rw-r--r--   1 root     root          184 Apr  7  2021 etc/cron.d/e2scrub_all
-rw-------   1 root     root          191 Nov 10  2022 etc/shadow
-rw-r--r--   1 root     root          685 Apr  7  2021 etc/e2scrub.conf
drwxrwxrwt   2 root     root            0 Jan 12  2020 tmp
drwxr-xr-x   2 root     root            0 Jan 12  2020 media
lrwxrwxrwx   1 root     root           11 Apr  7  2021 linuxrc -> bin/busybox
19987 blocks
```

So no, to the rpi-otp-private-key convenience script itself.
Yes, to /usr/bin/vcmailbox that does the real work.

closed 2024-02-06 by Siecje