raspberrypi/cmprovision #41: Rasberry Pi 4 Flash EEPROM firmware failes on RB4

knightofcarnage, 2024-02-08:
23:17:52 Error during preinstall. Return code 1. Script output:

===
Running pre-installation script 'Flash EEPROM firmware (stable/pieeprom-2023-01-11.bin)'
===
#!/bin/sh
set -e
curl --retry 10 --silent --show-error -g -o pieeprom.bin "http://172.20.0.1/uploads/pieeprom.bin"
echo "d9cd69334b26953eb66509ae751960b51aa1103786a736d9f63eefeb2e2cffc8 pieeprom.bin" | sha256sum -c
pieeprom.bin: OK
flashrom -p "linux_spi:dev=/dev/spidev0.0,spispeed=16000" -w "pieeprom.bin"
flashrom v1.1 on Linux 5.4.83-scriptexec (armv7l)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
l**inux_spi_init: failed to open /dev/spidev0.0: No such file or directory**
Error: Programmer initialization failed.

timg236, 2024-02-09:
This appears to be a custom flashing script so not sure why you are reporting the issue here.
You will will need resolve the SPI pin mux / audio issues and create an appropriate config.txt file to use flashrom on a Pi4 which is why we don’t support it.
You could look at the rpi-eeprom-update script for inspiration

knightofcarnage, 2024-02-09:
I am using a custom Openwrt image to be flashed, but I have not created any custom scripts.  It's a script that seems to have be created by ScriptExecuteController.php I only selected the newest EEPROM version in my RB4 project and changed BOOT_UART=0 to BOOT_UART=1.  If I don't select an EEPROM version everything works.  But I make a custom script based on your suggestion.

 Are there any forms for this project?  I like to learn more about the Lable feature.  Thank you for your response and time.

[image]

maxnet, 2024-02-09:
>This appears to be a custom flashing script so not sure why you are reporting the issue here.

As the TS already mentioned, it is OUR flashing script that gets inserted implicitly if the user selects an EEPROM image.
(when cmprovision was developed, flashrom was the only thing that worked properly in a network boot scenario. Network booting the recovery.bin stuff didn't work consistently at the time. Or as I recall the flashing part did work, but rebooting/network booting again afterwards did not work reliably).

As far as it not working on Pi 4.
EEPROM programming was written with CM4 in mind.
If you open /var/lib/cmprovision/scriptexecute/config.txt and change:

```
[cm4]
dtoverlay=dwc2
# For flashrom
dtparam=audio=off
dtoverlay=spi-gpio40-45
````

To:

```
[pi4]
dtoverlay=dwc2
# For flashrom
dtparam=audio=off
dtoverlay=spi-gpio40-45
```

It SHOULD work on Pi 4 as well.

knightofcarnage, 2024-02-09:
I will test out your suggestion. If this works them is will modify the project page to include a hardware target drop-down for RB4/CM4/RB5 and apply the change to config.txt when the project is activated.  My main use case is for CM4 and our custom carrier board, but we still sell products using RB4 and might be also using RB5.  Just trying to make everything automated and have consistency with EEPROM versions and configurations.   Also, need to test the automatic Lable feature with a Brady i3300.  Is there any additional documentation on this feature?  Thank you.

maxnet, 2024-02-09:
Don't need a drop down.
If you have [pi4] it applies to both cm4 and pi 4.

No support for Pi 5 at the moment.

Brady printers should work if they have the network module.
Just give the sample label a try, entering the printer's IP and FTP password in the settings.

knightofcarnage, 2024-02-11:
Changing to [pi4] worked, thank you.  Is there a form dedicated to this project? I have lots of questions about customizing pre-install scripts.