raspberrypi/cmprovision #42: Device can't connect to server

mortenboye, 2024-02-19:
I've been following the setup guide to install CM4 Provisioning on a Pi4 in order to flash CM4 on a custom board.

CM4 is connected via Ethernet to a (non managed) switch connected to the Pi4, as per the instructions, but the provisioning server never detects the CM4.

A project is created and an image is uploaded and ready.

In the dnsmasq log I see this:
```
Feb 19 10:40:20 mortrix-provisioning systemd[1]: Starting cmprovision-dnsmasq.service - cmprovision-dnsmasq...
Feb 19 10:40:20 mortrix-provisioning dnsmasq[10667]: dnsmasq: syntax check OK.
Feb 19 10:40:20 mortrix-provisioning dnsmasq[10673]: started, version 2.89 DNS disabled
Feb 19 10:40:20 mortrix-provisioning dnsmasq[10673]: compile time options: IPv6 GNU-getopt DBus no-UBus i18n IDN2 DHCP DHCPv6 no-Lua TFTP conntrack ipset nftset auth cryptohash DNSSEC loop-detect inotify dumpfile
Feb 19 10:40:20 mortrix-provisioning dnsmasq-dhcp[10673]: DHCP, IP range 172.20.0.2 -- 172.20.255.255, lease time 1h
Feb 19 10:40:20 mortrix-provisioning dnsmasq-tftp[10673]: TFTP root is /var/lib/cmprovision/scriptexecute
Feb 19 10:40:20 mortrix-provisioning systemd[1]: Started cmprovision-dnsmasq.service - cmprovision-dnsmasq.
Feb 19 10:58:31 mortrix-provisioning dnsmasq-dhcp[10673]: DHCP packet received on eth0 which has no address
Feb 19 10:58:39 mortrix-provisioning dnsmasq-dhcp[10673]: DHCP packet received on eth0 which has no address
Feb 19 10:58:55 mortrix-provisioning dnsmasq-dhcp[10673]: DHCP packet received on eth0 which has no address
Feb 19 10:59:33 mortrix-provisioning dnsmasq-dhcp[10673]: DHCP packet received on eth0 which has no address
Feb 19 10:59:41 mortrix-provisioning dnsmasq-dhcp[10673]: DHCP packet received on eth0 which has no address
Feb 19 10:59:57 mortrix-provisioning dnsmasq-dhcp[10673]: DHCP packet received on eth0 which has no address
Feb 19 11:00:39 mortrix-provisioning dnsmasq-dhcp[10673]: DHCP packet received on eth0 which has no address
...
```

With a new entry of `DHCP packet received on eth0 which has no address` every time the CM4 retries PXE boot.

`cat /etc/dhcpcd.conf`
gives me:
```  
interface eth0
static ip_address=172.20.0.1/16
```

`cat /var/lib/cmprovision/etc/dnsmasq.conf`
results in:
```
# No DNS
port=0

# tftp
enable-tftp
tftp-root=/var/lib/cmprovision/scriptexecute

# dhcp
dhcp-match=set:client_is_a_pi,97,0:52:50:69:34
dhcp-match=set:client_is_a_pi,97,0:34:69:50:52

log-dhcp
dhcp-range=172.20.0.2,172.20.255.255,255.255.0.0
pxe-service=tag:client_is_a_pi,0,"Raspberry Pi Boot"
dhcp-leasefile=/var/lib/cmprovision/etc/dnsmasq.leases
no-ping
```

And the CM4 keep trying to boot like this:
[image]

Any clues as to how to fix this? Did I miss something obvious?

maxnet, 2024-02-19:
>Instructions says "add ... to the bottom of the file", but the file was empty to begin with?

It's easiest if you start over with the previous RPI OS version, 64-bit edition.
Available with Imager under "Raspberry Pi OS (other)" -> "Raspberry Pi OS (Legacy, 64-bit)"

The new RPI OS version uses NetworkManager instead of dhcpcd, and needs new instructions to setup a static IP address.

mortenboye, 2024-02-19:
You were right. Reinstalled with the legacy version and now it works wonderfully. 

Thank you.

closed 2024-02-19 by mortenboye