raspberrypi/cmprovision #43: Download new firmware from GitHub.

knightofcarnage, 2024-02-19:
Downloading new firmware from GitHub is not working.  It returns firmware updated; the newest in the list is pieeprom-2023-01-11.bin. But there are 3 new images available: 

pieeprom-2023-05-11.bin
pieeprom-2024-01-18.bin
pieeprom-2024-01-22.bin

Is there a procedure to add the new images by hand?  Thanks -Mar

ric-osorio, 2024-03-04:
Fixed: chmod -R 777 /usr/lib/firmware/raspberrypi/bootloader-2711

patrickgroothuizen, 2024-03-06:
I'm facing the same issue. After downloading new firmware files from the web interface, I can find them all in the folder /usr/lib/firmware/raspberrypi/bootloader-2711/stable. Unfortunately, the web interface does not show them. Even not after a reboot of the Pi or cleaning the webbrowsers cache.

[image]

ric-osorio, 2024-03-06:
Try to add to /var/lib/cmprovision/storage/app/firmware/stable

patrickgroothuizen, 2024-03-06:
Thanks! Here the newly downloaded FW files are indeed not present.

[image]

ric-osorio, 2024-03-06:
What I did was to do a wget directly to this folder and it shows back in the project:

cd /var/lib/cmprovision/storage/app/firmware/stable
wget https://github.com/raspberrypi/rpi-eeprom/raw/master/firmware-2711/latest/pieeprom-2024-01-22.bin

patrickgroothuizen, 2024-03-06:
This works indeed 👍.

As an alternative, you can also copy the downloaded files manually to the right folder. 

sudo cp /usr/lib/firmware/raspberrypi/bootloader-2711/stable/pieeprom* /var/lib/cmprovision/storage/app/firmware/stable

Thanks again!

djr-spectrummfg, 2024-04-04:
The code in /app/Http/Livewire/Firmwares.php is looking for a directory in the downloaded zip file called "firmware", but this has since been changed to "firmware-2711" and "firmware-2712"

DaveMabri, 2024-09-16:
Please be aware that at least one version of the manually copied eeprom-firmware updates **does not work properly** on CM4 16GB & 32GB eMMC version:
[pieeprom-2024-07-30.bin](https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware-2711/latest/pieeprom-2024-07-30.bin)
Once written, the bootloader still booted the eMMC-located OS, but DID NOT run through the network boot provisioning server search procedure. It somehow skipped that step every time and went right to eMMC boot, although the boot order was set correctly for EEPROM settings with

> [all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
BOOT_ORDER=0xf56412
ENABLE_SELF_UPDATE=1

I did not further investigate, why that firmware update behaves this way, but reverted manually back to version pieeprom-2023-01-11, following the instructions from [how-update-raspberry-pi-compute-module-4-bootloader-eeprom](https://www.jeffgeerling.com/blog/2022/how-update-raspberry-pi-compute-module-4-bootloader-eeprom) 
This is especially frustrating if you already set up your device to only image flash from network, but then have to re-place the CM4 unto the CM4IO-Board, in order to get access to the microUSB-port. Maybe there is another way to fix a broken bootloader version via ethernet. 
I cannot confirm nor deny whether the other manually added bootloader firmware versions do function properly, as I did not try flashing those. I can only recommend the last official version [firmware-2711/latest/pieeprom-2023-01-11.bin](https://github.com/raspberrypi/rpi-eeprom/blob/d05f05c94fd7a1916942e56d7f486c142ae5661e/firmware-2711/latest/pieeprom-2023-01-11.bin).

timg236, 2024-09-16:
pieeprom-2024-07-30.bin is the generic firmware and will not contain the CM4 specific BOOT_ORDER.

You need to run update-pieeprom.sh in the recovery/ directory to apply the boot configuration.

See: https://github.com/raspberrypi/usbboot/commit/e7aefe3062f6c362afb4a78c6545c5b608ebcc22

We used to store output of update-pieeprom.sh but that was removed because it's unnecesary duplication of firmware files which can get out of sync

DaveMabri, 2024-09-16:
From my understanding of the CM Provisioning system, the EEPROM settings provided inside of the GUI for a project are applied after bootloader firmware update, 
I wrote those settings correctly, but they have not been applied after a complete image provisioning. Therefore, I was not able to re-network-boot the CM4 afterwards. 
I actually modified the instructions, too, when I tried the manual bootloader modifiication process. It succeeded with 
`cd recovery`
`nano boot.conf // & modify boot order`
`./update-pieeprom-sh`
`sudo ../rpiboot -d . -v | tee log`
This process was not the problem. 
The problem, I encountered, was that the normal provisioning procedure using only a provisioning server and CM4-to-provide connected via Ethernet with correctly modified EEPROM-settings DID NOT leave me with a flashed CM4 which scans for provisioning server and network booting if image is provided at startup.
Instead, I ended up with a CM4 which somehow skips that step or has not successfully modifed its boot order using the CM provisioning system. 
If it helps I could take a screenshot of the booting sequence log with a regularly functioning network-booting CM4 and the broken one with pieeprom-2024-07-30.bin. 

Maybe I did not understand your response, but I expected that the boot order would still be modifiable right during provisioning process with a newer bootloader firmware version as well. On the other hand, I did not read any commit messages about that stable version, so that it was maybe already mentioned, although I expected downwards compatibility.

timg236, 2024-09-16:
The best thing would be to change the config to BOOT_UART=1 and capture the UART log to see whether the DHCP / TFTP is succeeding. 
For reference this is the log from the CM4 on my desk

```
  0.07 RPi: BOOTSYS release VERSION:5f18ffb6 DATE: 2024/07/30 TIME: 14:02:35
  0.11 BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1722344555 serial f0560721 boardrev c03141 stc 511954
  0.27 PM_RSTS: 0x00001000
  0.27 part 00000000 reset_info 00000000
  0.31 uSD voltage 3.3V
  0.56 Initialising SDRAM 'Samsung' 16Gb x2 total-size: 32 Gbit 3200
  0.59 DDR 3200 1 0 32 152
  2.28 OTP boardrev c03141 bootrom 48b0 48b0
  2.29 Customer key hash 0000000000000000000000000000000000000000000000000000000000000000
  2.37 VC-JTAG unlocked
  2.12 RPi: BOOTLOADER release VERSION:5f18ffb6 DATE: 2024/07/30 TIME: 14:02:35
  2.17 BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1722344555 serial f0560721 boardrev c03141 stc 2417058
  2.52 Boot mode: NETWORK (02) order f5641
  2.53 GENET: RESET_PHY 1
  2.56 CTL 1140 PHY ID 600d 84a2
  2.58 MII_CONTROL 1140
  2.61 MII APD 0021 SCTL3 0021 IDDQ 0000

  2.40 NETWORK: d8:3a:dd:05:fd:51 wait for link TFTP: 0.0.0.0
  5.21 LINK STATUS: speed: 1000 full duplex
  5.22 Link ready
  5.24 GENET START: 64 16 32
  5.38 GENET: UMAC_START 0xd83add05 0xfd510000
  5.39 RX: 0 IP: 0 IPV4: 0 MAC: 0 UDP: 0 UDP RECV: 0 IP_CSUM_ERR: 0 UDP_CSUM_ERR: 0
  7.84 HDMI0 edid block 0 offset 0
  7.84 00ffffffffffff0010accfa142493530
  7.88 0e210103803c2278ea5095a8544ea526
  7.92 0f5054a54b00714f8180a9c0a940d1c0
  7.95 e1000101010108e80030f2705a80b058
  7.99 8a0055502100001e000000ff00393636
  7.02 344d44330a2020202020000000fc0044
  7.06 454c4c20533237323251430a000000fd
  7.09 00184c0a893c000a2020202020200143
  7.25 HDMI0 edid block 1 offset 128
  7.26 020346f1556101020304050607101112
  7.29 131415161f20215d5e5f230907078301
  7.33 00006d030c0020003844200060030201
  7.36 67d85dc401788001e40f010000681a00
  7.40 000101283ce6565e00a0a0a029503020
  7.44 350055502100001a0000000000000000
  7.47 00000000000000000000000000000000
  7.51 00000000000000000000000000000009
  7.54 HDMI0: best-mode 1 (limit 1) 1280x720 60 Hz CEA modes fe007f8003000000000000e002000000 extensions 1
 13.40 DHCP src: 86:98:d3:31:8e:ac 10.3.31.1
 13.41 YI_ADDR 10.3.31.35
 13.43 SI_ADDR 10.3.31.1
 13.46 OPTIONS:-
        op: 53 len:   1 DHCP recv OFFER (2) expect OFFER 13.51
        op: 54 len:   4 10.3.31.1 13.54
        op: 51 len:   4  13.57
        op: 58 len:   4  13.59
        op: 59 len:   4  13.61
        op:  1 len:   4 255.255.255.0 13.65
        op: 28 len:   4 10.3.31.255 13.68
        op:  3 len:   4 10.3.31.1 13.71
 13.86 DHCP src: 52:54:00:ae:0d:66 10.3.31.18
 13.87 OPTIONS:-
        op: 53 len:   1 DHCP recv OFFER (2) expect OFFER 13.93
        op: 54 len:   4 10.3.31.18 13.96
        op: 60 len:   9  13.98
        op: 97 len:  17  13.01
        op: 43 len:  32
                PXE op:  6 len:   1
                PXE op: 10 len:   4
                PXE op:  9 len:  20
00000000  52 61 73 70 62 65 72 72  79 20 50 69 20 42 6f 6f |Raspberry Pi Boo 13.16 |
[... 37 lines cut ...]
 13.70 TFTP_GET: 52:54:00:ae:0d:66 10.3.31.18 f0560721/recover4.elf

 13.98 RX: 130 IP: 0 IPV4: 77 MAC: 27 UDP: 27 UDP RECV: 7 IP_CSUM_ERR: 0 UDP_CSUM_ERR: 0
 13.04 TFTP 1: file /tftpboot/f0560721/recover4.elf not found for 10.3.31.35
 13.10 RX: 135 IP: 0 IPV4: 78 MAC: 28 UDP: 28 UDP RECV: 8 IP_CSUM_ERR: 0 UDP_CSUM_ERR: 0
 13.18 Read f0560721/recover4.elf failed
 13.21 TFTP_GET: 52:54:00:ae:0d:66 10.3.31.18 f0560721/recovery.elf

 13.27 RX: 135 IP: 0 IPV4: 78 MAC: 28 UDP: 28 UDP RECV: 8 IP_CSUM_ERR: 0 UDP_CSUM_ERR: 0
 13.36 TFTP 1: file /tftpboot/f0560721/recovery.elf not found for 10.3.31.35
 13.42 RX: 137 IP: 0 IPV4: 80 MAC: 29 UDP: 29 UDP RECV: 9 IP_CSUM_ERR: 0 UDP_CSUM_ERR: 0
 13.50 Read f0560721/recovery.elf failed
 13.54 TFTP_GET: 52:54:00:ae:0d:66 10.3.31.18 f0560721/start4.elf

 13.59 RX: 137 IP: 0 IPV4: 80 MAC: 29 UDP: 29 UDP RECV: 9 IP_CSUM_ERR: 0 UDP_CSUM_ERR: 0
 14.43 TFTP: complete 2257216
 14.43 RX: 142 IP: 0 IPV4: 83 MAC: 32 UDP: 32 UDP RECV: 12 IP_CSUM_ERR: 0 UDP_CSUM_ERR: 0
 14.51 Read f0560721/start4.elf bytes  2257216 hnd 0x0
 14.80 TFTP_GET: 52:54:00:ae:0d:66 10.3.31.18 f0560721/fixup4.dat

 14.83 RX: 142 IP: 0 IPV4: 83 MAC: 32 UDP: 32 UDP RECV: 12 IP_CSUM_ERR: 0 UDP_CSUM_ERR: 0
 14.94 TFTP: complete 5436
 14.94 RX: 147 IP: 0 IPV4: 88 MAC: 37 UDP: 37 UDP RECV: 17 IP_CSUM_ERR: 0 UDP_CSUM_ERR: 0

```