raspberrypi/cmprovision #44: issue while provisioning a CM4 module

Smeets-Marc, 2024-02-27:
hi,

i'm currently trying to use the provisioning software to load an image onto a new compute 4 board.
The software sees the compute 4  and starts working. 
I see the following happening on the dashboard:
12:35:21 Provisioning started. starting preinstalled scripts
12:35:22 preinstall script complete. starting to write image.
12:36:20 error during dd. return code 1. script output
2403438+59826 records in
1187+1 records out
I was trying to upload the Raspios bullseye-armhf.img.xz image . It allways ends with the same error. 
What's going wrong?

maxnet, 2024-02-27:
Connect a HDMI display to the board you are provisioning, and look at the screen to see if it has more information.

Smeets-Marc, 2024-03-01:
the upload allways failed due to a corrupted XZ file. 
I have decompressed the XZ file and compressed with Gzip. Now it uploads without a problem. 
It isn't the image itself because decompress and compress in another format solved my problem.
It looks something goes wrong while decompressing the XZ file during provisioning. 
Thanks for the tip about the HDMI screen it really helped.