raspberrypi/cmprovision #46: How to boot installed image immediately after provisioning

djr-spectrummfg, 2024-04-04:
I would like the CM4 system to boot the newly-installed image immediately after provisioning is complete. I have attempted to add a postinstall script that calls the `reboot` command, however this does not appear to work.

maxnet, 2024-04-04:
"reboot" only sends a signal to an init process such as systemd to nudge it to reboot.
In environments in which there is no init process, you may need to make sure the reboot() syscall is invoked instead.

```
sync
reboot -f
```

Do keep in mind rebooting to your OS is not possible if USB booting, only when Ethernet booting.

djr-spectrummfg, 2024-04-04:
Thanks, using `reboot -f` is working (even though the CM4 was connected to USB to boot)

maxnet, 2024-04-04:
Meant rpiboot USB booting, not USB MSD ("usb stick") booting.
If jumper is set for rpiboot, it will not do a normal boot, making reboot to normal OS difficult.

djr-spectrummfg, 2024-04-04:
Right, I was using rpiboot but I have a momentary switch rather than a jumper.

closed 2024-04-05 by djr-spectrummfg

djr-spectrummfg, 2024-04-05:
One remaining issue is that while using `sync` + `reboot -f` works, it also results in the CM remaining with "Provisioning complete" as "No" in the web interface, which is not ideal. 

However this issue can be avoided by checking the **Run in background** option and adding delays to the script (which I already had to be sure the sync was complete).

I ended up going with the script below:
```sh
#!/bin/sh
set -e
sync
sleep 2
sync
sleep 2
reboot -f
```

maxnet, 2024-04-05:
```
One remaining issue is that while using sync + reboot -f works, it also results in the CM remaining with "Provisioning complete" as "No" in the web interface, which is not ideal.
```

If there is stuff left to do on first boot, you typically let your first boot script phone home completion.

How such things are done best is very dependent on the OS you are installing though.
E.g. if you are using Ubuntu Core or Ubuntu Server, and let cmprovision generate the cloud-init user-data file (with a little bit of extra code, support for that is not in the stock cmprovision version), you typically add a section to your template among the lines of:

```
phone_home:
  url: http://{{ $server }}/scriptexecute?serial={{ $cm->serial }}&alldone=1&verify={{ $project->verify }}
  tries: 10
```