raspberrypi/cmprovision #49: provisioner on bookworm

knachte, 2024-04-16:
I've set up a provisioner on a bookworm install and am seeing problems with the network settings, both in the manual which is no longer up to date and after installing the provisioner, with files like /etc/dhcpcd.conf no longer being he correct ones to use. 

Are there plans to upgrade the provisioner/documentation to work with bookworm?

synaption, 2024-05-30:
what are you using instead of /etc/dhcpcd.conf?

synaption, 2024-06-28:
I did it with /etc/systemd/network/eth0.network
```
[Match]
Name=eth0

[Network]
Address=172.20.0.1/16
```

synaption, 2024-06-28:
I did this:
sudo nano /etc/network/interfaces.d/eth0
```
auto eth0
iface eth0 inet static
    address 172.20.0.1
    netmask 255.255.0.0
```

closed 2024-08-29 by knachte

magicsmoke, 2024-09-19:
Well I just wasted half a day on this. Suspecting originally a hardware issue.

Updated documentation would be have been very appreciated.