raspberrypi/cmprovision #53: Image is not a valid disk image. on one provisioner but not another?

synaption, 2024-07-01:
I am getting this error message on one device but not another.
```Image is not a valid disk image. Uncompressed size not dividable by sector size of 512 bytes.```
They are trying to upload the same image.  
One device is running the latest rasberry pi os bookworm, the other is running an old version of buster.  
I'm not sure what else could be different that causes this error.  Should I start from scratch?

synaption, 2024-07-01:
I tried "Verify that image was written correctly" set to on and off on both with no difference.  I have now zero padded the image which seems to work.  ```truncate -s %512  sdcard.img.bz2```

maxnet, 2024-07-01:
If you are going to zero pad, do make sure you are padding the uncompressed image, and not a .bz2