raspberrypi/cmprovision #56: Documentation on label printing?

aacos-uy, 2024-09-24:
Sorry if this is posted somewhere but couldn't find more info.
I'm more of a hardware guy, right now I'm trying to use a Brother Ptouch printer to print labels after provisoning. So far the printer is working on the rpi 4 server (used this guide: https://github.com/HenrikBengtsson/brother-ptouch-label-printer-on-linux) but I can't figure the options on "labels/add label" to output.
Is there a detailed document on how to use this?
Thanks again and sorry for the trouble

maxnet, 2024-09-25:
The software assumes you are using a label printer that can be fed a text document with markup elements, and that the printer then does its thing.
E.g. you can draft a text template in the ZPL markup language if you are using a Zebra printer.
Or one in the JScript markup language for Brady/CAB printers.

In case you are using a consumer grade printer that does not understand any markup language, but just wants to receive pixel raster data, you will need to use an intermediate program that generates that.
One way is to draft the template in HTML in cmprovision, and configure that it pipes the result to "dompdf", which converts the HTML to PDF.
PDF is a file format that can then be printed by lpr, if your printer has a CUPS driver.

See also:
https://github.com/raspberrypi/cmprovision/issues/25#issuecomment-1602838683
https://github.com/trialinfo/ptouch-driver

aacos-uy, 2024-09-25:
Thanks! Will try to get a Zebra printer (that's what's available locally) as it will make things much simpler.

closed 2024-09-25 by aacos-uy