raspberrypi/cmprovision #61: cmprovision working on cm5

jcaptitude, 2024-12-13:
Hi,

I've just got some CM5s and installed the cmprovision on cm5.  The static ip was setup following this [issue](https://github.com/raspberrypi/cmprovision/issues/57)

But it seems that it cannot detect any CM5 connected to it.

The dnsmasq log only shows 'no address range available for DHCP request via wlan0',
And the rpiboot log stops at 'Waiting for BCM2835/6/7/2711/2712...'

So I'm wondering if cmprovision is able to work on CM5?

timg236, 2024-12-16:
I believe rpi-sb-provisioner is the preferred tool on CM5 and should a superset of the cmprovsion features (although some are implemented differently)
https://github.com/raspberrypi/rpi-sb-provisioner

Re "rpiboot" please can you make sure that you are using the very latest APT release of rpiboot. Early versions of the bootloader (including bootfiles.bin) won't boot on a CM5 or Pi500.

jayjayseal, 2025-01-20:
Add: 
dhcp-match=set:client_is_a_pi,97,0:35:69:50:52

To DNSmasq :-) and it will attempt to boot.

pirvu17, 2025-01-20:
I had the same problem, and after adding the line:
dhcp-match=set:client_is_a_pi,97,0:35:69:50:52
to the DNSmasq configuration, the device attempted to boot. However, after copying the data from the /boot partition of a compatible Raspberry Pi CM5 image, I encountered the error:

file /var/lib/cmprovision/scriptexecute/71da75ea/armstub8-2712.bin not found

Full log:

-- Journal begins at Tue 2024-10-22 15:54:18 EEST, ends at Mon 2025-01-20 19:42:16 EET. --
Jan 20 19:26:20 pi3 systemd[1]: Starting cmprovision-dnsmasq...
Jan 20 19:26:20 pi3 dnsmasq[474]: dnsmasq: syntax check OK.
Jan 20 19:26:20 pi3 dnsmasq[491]: started, version 2.85 DNS disabled
Jan 20 19:26:20 pi3 dnsmasq[491]: compile time options: IPv6 GNU-getopt DBus no-UBus i18n IDN2 DHCP DHCPv6 no-Lua TFTP conntrack ipset auth cryptohash DNSSEC loop-detect inotify dumpfile
Jan 20 19:26:20 pi3 dnsmasq-dhcp[491]: DHCP, IP range 172.20.0.2 -- 172.20.255.255, lease time 1h
Jan 20 19:26:20 pi3 dnsmasq-tftp[491]: TFTP root is /var/lib/cmprovision/scriptexecute
Jan 20 19:26:20 pi3 systemd[1]: Started cmprovision-dnsmasq.
Jan 20 19:41:44 pi3 dnsmasq-dhcp[491]: 1076043725 available DHCP range: 172.20.0.2 -- 172.20.255.255
Jan 20 19:41:44 pi3 dnsmasq-dhcp[491]: 1076043725 vendor class: PXEClient:Arch:00000:UNDI:002001
Jan 20 19:41:44 pi3 dnsmasq-dhcp[491]: 1076043725 DHCPDISCOVER(eth0) 2c:cf:67:c4:ac:a4
Jan 20 19:41:44 pi3 dnsmasq-dhcp[491]: 1076043725 tags: client_is_a_pi, eth0
Jan 20 19:41:44 pi3 dnsmasq-dhcp[491]: 1076043725 DHCPOFFER(eth0) 172.20.57.156 2c:cf:67:c4:ac:a4
Jan 20 19:41:44 pi3 dnsmasq-dhcp[491]: 1076043725 requested options: 1:netmask, 3:router, 43:vendor-encap, 60:vendor-class,
Jan 20 19:41:44 pi3 dnsmasq-dhcp[491]: 1076043725 requested options: 66:tftp-server, 67:bootfile-name, 128, 129,
Jan 20 19:41:44 pi3 dnsmasq-dhcp[491]: 1076043725 requested options: 130, 131, 132, 133, 134, 135
Jan 20 19:41:44 pi3 dnsmasq-dhcp[491]: 1076043725 next server: 172.20.0.1
Jan 20 19:41:44 pi3 dnsmasq-dhcp[491]: 1076043725 sent size:  1 option: 53 message-type  2
Jan 20 19:41:44 pi3 dnsmasq-dhcp[491]: 1076043725 sent size:  4 option: 54 server-identifier  172.20.0.1
Jan 20 19:41:44 pi3 dnsmasq-dhcp[491]: 1076043725 sent size:  4 option: 51 lease-time  1h
Jan 20 19:41:44 pi3 dnsmasq-dhcp[491]: 1076043725 sent size:  4 option: 58 T1  30m
Jan 20 19:41:44 pi3 dnsmasq-dhcp[491]: 1076043725 sent size:  4 option: 59 T2  52m30s
Jan 20 19:41:44 pi3 dnsmasq-dhcp[491]: 1076043725 sent size:  4 option:  1 netmask  255.255.0.0
Jan 20 19:41:44 pi3 dnsmasq-dhcp[491]: 1076043725 sent size:  4 option: 28 broadcast  172.20.255.255
Jan 20 19:41:44 pi3 dnsmasq-dhcp[491]: 1076043725 sent size:  4 option:  3 router  172.20.0.1
Jan 20 19:41:44 pi3 dnsmasq-dhcp[491]: 1076043725 sent size:  9 option: 60 vendor-class  50:58:45:43:6c:69:65:6e:74
Jan 20 19:41:44 pi3 dnsmasq-dhcp[491]: 1076043725 sent size: 17 option: 97 client-machine-id  00:35:69:50:52:80:41:b0:00:67:c4:ac:a4:ea...
Jan 20 19:41:44 pi3 dnsmasq-dhcp[491]: 1076043725 sent size: 32 option: 43 vendor-encap  06:01:03:0a:04:00:50:58:45:09:14:00:00:11...
Jan 20 19:41:44 pi3 dnsmasq-dhcp[491]: 1076043725 available DHCP range: 172.20.0.2 -- 172.20.255.255
Jan 20 19:41:44 pi3 dnsmasq-dhcp[491]: 1076043725 vendor class: PXEClient:Arch:00000:UNDI:002001
Jan 20 19:41:44 pi3 dnsmasq-dhcp[491]: 1076043725 DHCPREQUEST(eth0) 172.20.57.156 2c:cf:67:c4:ac:a4
Jan 20 19:41:44 pi3 dnsmasq-dhcp[491]: 1076043725 tags: client_is_a_pi, eth0
Jan 20 19:41:44 pi3 dnsmasq-dhcp[491]: 1076043725 DHCPACK(eth0) 172.20.57.156 2c:cf:67:c4:ac:a4
Jan 20 19:41:44 pi3 dnsmasq-dhcp[491]: 1076043725 next server: 172.20.0.1
Jan 20 19:41:44 pi3 dnsmasq-dhcp[491]: 1076043725 sent size:  1 option: 53 message-type  5
Jan 20 19:41:44 pi3 dnsmasq-dhcp[491]: 1076043725 sent size:  4 option: 54 server-identifier  172.20.0.1
Jan 20 19:41:44 pi3 dnsmasq-dhcp[491]: 1076043725 sent size:  4 option: 51 lease-time  1h
Jan 20 19:41:44 pi3 dnsmasq-dhcp[491]: 1076043725 sent size:  4 option: 58 T1  30m
Jan 20 19:41:44 pi3 dnsmasq-dhcp[491]: 1076043725 sent size:  4 option: 59 T2  52m30s
Jan 20 19:41:44 pi3 dnsmasq-dhcp[491]: 1076043725 sent size:  4 option:  1 netmask  255.255.0.0
Jan 20 19:41:44 pi3 dnsmasq-dhcp[491]: 1076043725 sent size:  4 option: 28 broadcast  172.20.255.255
Jan 20 19:41:44 pi3 dnsmasq-dhcp[491]: 1076043725 sent size:  4 option:  3 router  172.20.0.1
Jan 20 19:41:44 pi3 dnsmasq-tftp[491]: error 0 Early terminate received from 172.20.57.156
Jan 20 19:41:44 pi3 dnsmasq-tftp[491]: sent /var/lib/cmprovision/scriptexecute/71da75ea/config.txt to 172.20.57.156
Jan 20 19:41:44 pi3 dnsmasq-tftp[491]: sent /var/lib/cmprovision/scriptexecute/71da75ea/config.txt to 172.20.57.156
Jan 20 19:41:44 pi3 dnsmasq-tftp[491]: file /var/lib/cmprovision/scriptexecute/71da75ea/pieeprom.sig not found
Jan 20 19:41:44 pi3 dnsmasq-tftp[491]: sent /var/lib/cmprovision/scriptexecute/71da75ea/bcm2712-rpi-5-b.dtb to 172.20.57.156
Jan 20 19:41:44 pi3 dnsmasq-tftp[491]: sent /var/lib/cmprovision/scriptexecute/71da75ea//config.txt to 172.20.57.156
Jan 20 19:41:44 pi3 dnsmasq-tftp[491]: sent /var/lib/cmprovision/scriptexecute/71da75ea//config.txt to 172.20.57.156
Jan 20 19:41:45 pi3 dnsmasq-tftp[491]: error 0 Early terminate received from 172.20.57.156
Jan 20 19:41:45 pi3 dnsmasq-tftp[491]: sent /var/lib/cmprovision/scriptexecute/71da75ea/kernel_2712.img to 172.20.57.156
[... cut ...]
Jan 20 19:41:50 pi3 dnsmasq-tftp[491]: sent /var/lib/cmprovision/scriptexecute/71da75ea//config.txt to 172.20.57.156
Jan 20 19:41:50 pi3 dnsmasq-tftp[491]: sent /var/lib/cmprovision/scriptexecute/71da75ea//overlays/dwc2.dtbo to 172.20.57.156
Jan 20 19:41:50 pi3 dnsmasq-tftp[491]: sent /var/lib/cmprovision/scriptexecute/71da75ea//cmdline.txt to 172.20.57.156
Jan 20 19:41:50 pi3 dnsmasq-tftp[491]: file /var/lib/cmprovision/scriptexecute/71da75ea/armstub8-2712.bin not found
Jan 20 19:41:54 pi3 dnsmasq-tftp[491]: sent /var/lib/cmprovision/scriptexecute/71da75ea/kernel_2712.img to 172.20.57.156

jayjayseal, 2025-01-23:
Well you need to compile a new kernel for the 2712 chip too.... and i managed to compile one that is working and booting. Though i'm stuck at the Welcome to buildroot prompt myself. It seems to be working but it's not starting the copy process. 

I'm probably compiling wrong here

maxnet, 2025-01-30:
>I'm probably compiling wrong here
>Using the default raspberry5_defconfig (it is a rpi5 not a CM though. for CM5 there is no defconfig that i have found)

You normally would compile things using this buildroot project: https://github.com/raspberrypi/scriptexecutor/

The scripting merges two kernel configuration files.

It expects you to supply a bare bones kernel configuration file without any modules.
For CM4 that was this one: https://github.com/raspberrypi/scriptexecutor/blob/master/scriptexecute/board/kernelconfig-cm4.armv7
Which was originally created by running: "cat bcm2711_defconfig |grep -v =m"

And on top of that there is a configuration file with the extras cmprovision expects to be statically compiled into the kernel (not as module): https://github.com/raspberrypi/scriptexecutor/blob/master/scriptexecute/board/kernelconfig-scriptexecute.fragment
This are options necessary to support GPIO (to test for jumpers set, used by the labelling mechanism), support for flashing EEPROM firmware with flashrom, etc.
As well as the alternative dwc2 USB driver (necessary to allow communication by USB Ethernet gadget instead of Ethernet).

You will also need to change the buildroot configuration to tell it to build the cm5 DTB file from kernel source: https://github.com/raspberrypi/scriptexecutor/blob/master/scriptexecute/configs/scriptexecute_cm4_defconfig#L21

jayjayseal, 2025-01-31:
This is one i got from Buildroot

```
BR2_aarch64=y
BR2_cortex_a76=y
BR2_ARM_FPU_VFPV4=y

BR2_TOOLCHAIN_BUILDROOT_CXX=y

BR2_SYSTEM_DHCP="eth0"

# Linux headers same as kernel, a 6.1 series
BR2_PACKAGE_HOST_LINUX_HEADERS_CUSTOM_6_1=y

BR2_LINUX_KERNEL=y
BR2_LINUX_KERNEL_CUSTOM_TARBALL=y
BR2_LINUX_KERNEL_CUSTOM_TARBALL_LOCATION="$(call github,raspberrypi,linux,17f135b742c4edb340afb365873c3a574f7e16cb)/linux-17f135b742c4edb340afb365873c3a574f7e16cb.tar.gz"
BR2_LINUX_KERNEL_DEFCONFIG="bcm2712"

# Build the DTB from the kernel sources
BR2_LINUX_KERNEL_DTS_SUPPORT=y
BR2_LINUX_KERNEL_INTREE_DTS_NAME="broadcom/bcm2712-rpi-5-b"

BR2_LINUX_KERNEL_NEEDS_HOST_OPENSSL=y

BR2_PACKAGE_RPI_FIRMWARE=y
BR2_PACKAGE_RPI_FIRMWARE_VARIANT_PI4=y
BR2_PACKAGE_RPI_FIRMWARE_CONFIG_FILE="board/raspberrypi5/config_5.txt"

# Required tools to create the SD image
BR2_PACKAGE_HOST_DOSFSTOOLS=y
BR2_PACKAGE_HOST_GENIMAGE=y
BR2_PACKAGE_HOST_MTOOLS=y

# Filesystem / image
BR2_TARGET_ROOTFS_EXT2=y
BR2_TARGET_ROOTFS_EXT2_4=y
BR2_TARGET_ROOTFS_EXT2_SIZE="120M"
# BR2_TARGET_ROOTFS_TAR is not set
BR2_ROOTFS_POST_BUILD_SCRIPT="board/raspberrypi5/post-build.sh"
BR2_ROOTFS_POST_IMAGE_SCRIPT="board/raspberrypi5/post-image.sh"

# Enable compressed kernel module support
BR2_PACKAGE_BUSYBOX_SHOW_OTHERS=y
BR2_PACKAGE_KMOD_TOOLS=y
BR2_PACKAGE_XZ=y
BR2_PACKAGE_KMOD=y
BR2_PACKAGE_HOST_KMOD_XZ=y
```

maxnet, 2025-01-31:
The buildroot defconfigs you seem to be using, will give you an empty buildroot operating system project, without any software packages installed.

You really want to use the existing  scriptexecute_cm4_defconfig configuration as base instead and only change the kernel version, kernel configuration, dtb and architecture.

Upgrading buildroot to a later version itself shouldn't be a problem, as our changes are out-of-tree. They live in different folders than buildroot itself.

jayjayseal, 2025-02-06:
Hi, immediately get push back to the Buildroot login. When i start the init.d script manually i get a VCHI Initialized failed error. Now this only gets the temperature of the CPU. But you can get this other ways too
Changed 

`	TEMP=`vcgencmd measure_temp`
	TEMP=${TEMP:5}`

to

`	TEMP=`cat /sys/class/thermal/thermal_zone0/temp`
	TEMP=$((TEMP/1000))`

But when i change this (and on the server in the script it fetches as well)...... The script will then run and it starts to copy but has no output to the screen when it starts at boot with init.d.... when it's run manually from the command line it does have an output. So I'm missing a setting i guess :)

`BR2_aarch64=y
BR2_cortex_a76=y
BR2_PACKAGE_HOST_LINUX_HEADERS_CUSTOM_6_6=y
BR2_TOOLCHAIN_BUILDROOT_CXX=y
BR2_GLOBAL_PATCH_DIR="board/raspberrypi/patches"
BR2_DOWNLOAD_FORCE_CHECK_HASHES=y
BR2_TARGET_GENERIC_HOSTNAME="scriptexecute"
BR2_TARGET_GENERIC_ROOT_PASSWD="raspberry"
BR2_TARGET_GENERIC_GETTY_PORT="tty1"
BR2_TARGET_GENERIC_GETTY_BAUDRATE_115200=y
BR2_SYSTEM_DHCP="eth0"
BR2_ROOTFS_OVERLAY="$(BR2_EXTERNAL_SCRIPTEXECUTE_PATH)/board/overlay/"
BR2_LINUX_KERNEL=y
BR2_LINUX_KERNEL_CUSTOM_TARBALL=y
BR2_LINUX_KERNEL_CUSTOM_TARBALL_LOCATION="$(call github,raspberrypi,linux,576cc10e1ed50a9eacffc7a05c796051d7343ea4)/linux-576cc10e1ed50a9eacffc7a05c796051d7343ea4.tar.gz"
BR2_LINUX_KERNEL_DEFCONFIG="bcm2712"
BR2_LINUX_KERNEL_CONFIG_FRAGMENT_FILES="$(BR2_EXTERNAL_SCRIPTEXECUTE_PATH)/board/kernelconfig-scriptexecute.fragment"
BR2_LINUX_KERNEL_DTS_SUPPORT=y
BR2_LINUX_KERNEL_INTREE_DTS_NAME="broadcom/bcm2712-rpi-5-b broadcom/bcm2712d0-rpi-5-b overlays/dwc2-overlay overlays/spi-gpio40-45-overlay"
BR2_LINUX_KERNEL_NEEDS_HOST_OPENSSL=y
BR2_LINUX_KERNEL_NEEDS_HOST_LIBELF=y
BR2_PACKAGE_BUSYBOX_SHOW_OTHERS=y
BR2_PACKAGE_XZ=y
BR2_PACKAGE_E2FSPROGS=y
# BR2_PACKAGE_E2FSPROGS_FSCK is not set
BR2_PACKAGE_E2FSPROGS_RESIZE2FS=y
BR2_PACKAGE_MMC_UTILS=y
BR2_PACKAGE_RPI_FIRMWARE=y
# BR2_PACKAGE_RPI_FIRMWARE_INSTALL_DTB_OVERLAYS is not set
BR2_PACKAGE_FLASHROM=y
BR2_PACKAGE_PARTED=y
BR2_PACKAGE_RPI_USERLAND=y
BR2_PACKAGE_LIBARCHIVE=y
BR2_PACKAGE_LIBARCHIVE_BSDTAR=y
BR2_PACKAGE_ZLIB=y
BR2_PACKAGE_LIBSYSFS=y
BR2_PACKAGE_JITTERENTROPY_LIBRARY=y
BR2_PACKAGE_LIBGPIOD2=y
BR2_PACKAGE_LIBGPIOD2_TOOLS=y
BR2_PACKAGE_LIBCURL=y
BR2_PACKAGE_LIBCURL_CURL=y
# BR2_PACKAGE_LIBCURL_PROXY_SUPPORT is not set
# BR2_PACKAGE_LIBCURL_COOKIES_SUPPORT is not set
# BR2_PACKAGE_LIBCURL_EXTRA_PROTOCOLS_FEATURES is not set
BR2_PACKAGE_UTIL_LINUX_BINARIES=y
BR2_PACKAGE_NANO=y
BR2_TARGET_ROOTFS_CPIO=y
BR2_TARGET_ROOTFS_CPIO_XZ=y
# BR2_TARGET_ROOTFS_TAR is not set
BR2_PACKAGE_RPI_PINCTRL=y
`

maxnet, 2025-02-06:
You do have the default cmprovision cmdline.txt ?
If tgere is no cmdline the kernel may default to a builtin one that may include an option to send output to serial console.
Can also try setting something like console=tty1 explicitly.

jayjayseal, 2025-02-07:
It was that!

i added 
console=tty1 before readjumper script="...." and it started to work!

Now back to trying to see if it all works again :)

Dude90, 2025-04-11:
Hey!

I am currently facing the same problem: I would be happy to use the cmprovision tool for the CM5.

@jayjayseal Is your system working? Can I see/find your changes somewhere?
@maxnet Are there plans to update the official cmprovision tool with CM5 support?

maxnet, 2025-04-11:
>@maxnet Are there plans to add CM5 support to the official 'cmprovision' tool?

Not my call to make.
While I hang around as I like to see stuff I once write to be used, I am no longer officially involved.

Dude90, 2025-04-15:
@maxnet Thank you for your answer. I forwarded the question directly to the Raspberry Pi team [here](https://forums.raspberrypi.com/viewtopic.php?t=386712). Unfortunately no answer yet.

In the meantime I tried to add CM5 support to the _cmprovision_ tool myself, using the answers above. Unfortunately I was not successful. In following I am shortly describing my steps. Maybe someone can have a look at it and help get the CM5 provisioning to work.

1. Added "dhcp-match=set:client_is_a_pi,97,0:35:69:50:52" to '/var/lib/cmprovision/etc/dnsmasq.conf' -> CM5 on a CM5IO is recognized correctly by the cmprovision tool, gets an IP and NFS boot is initialized but fails
2. Replaced "buildroot-2019.11.1.tar.gz" with the current version "buildroot-2025.02.tar.gz" and updated the version in 'build.sh'
3. Created a "kernelconfig-cm5.armv7" by executing "cat bcm2712_defconfig | grep -v =m" on the ARM64 BCM2712 config for Kernel 6.6 (taken from [here](https://github.com/raspberrypi/linux/blob/rpi-6.6.y/arch/arm64/configs/bcm2712_defconfig)). Placed "kernelconfig-cm5.armv7" in '/scriptexecute/boards'
4. Changed architecture, kernel, config-file and dts in "scriptexecutor/configs/scriptexecute_cm4_defconfig" (final defconfig below)
5. Executed './build.sh' -> Fails (expected)
6. Executed 'make -C buildroot-2025.02 BR2_EXTERNAL="$PWD/scriptexecute" menuconfig'. Checked/Changed architecture, processor-type (A76), headers (6.6), kernel, etc. (final defconfig below) + unchecked all legacy options
Resulting 'scriptexecute_cm4_defconfig'
`BR2_aarch64=y
BR2_cortex_a76=y
BR2_ARM_FPU_VFPV2=y
BR2_TOOLCHAIN_BUILDROOT_UCLIBC=y
BR2_PACKAGE_HOST_LINUX_HEADERS_CUSTOM_6_6=y
BR2_TOOLCHAIN_BUILDROOT_LOCALE=y
# BR2_UCLIBC_INSTALL_UTILS is not set
BR2_TOOLCHAIN_BUILDROOT_CXX=y
BR2_TARGET_GENERIC_HOSTNAME="scriptexecute"
BR2_TARGET_GENERIC_ROOT_PASSWD="raspberry"
BR2_TARGET_GENERIC_GETTY_PORT="tty1"
BR2_TARGET_GENERIC_GETTY_BAUDRATE_115200=y
BR2_ROOTFS_OVERLAY="$(BR2_EXTERNAL_SCRIPTEXECUTE_PATH)/board/overlay/"
BR2_LINUX_KERNEL=y
BR2_LINUX_KERNEL_CUSTOM_GIT=y
BR2_LINUX_KERNEL_CUSTOM_REPO_URL="https://github.com/raspberrypi/linux.git"
BR2_LINUX_KERNEL_CUSTOM_REPO_VERSION="rpi-6.6.y"
BR2_LINUX_KERNEL_USE_CUSTOM_CONFIG=y
BR2_LINUX_KERNEL_CUSTOM_CONFIG_FILE="$(BR2_EXTERNAL_SCRIPTEXECUTE_PATH)/board/kernelconfig-cm5.armv7"
BR2_LINUX_KERNEL_CONFIG_FRAGMENT_FILES="$(BR2_EXTERNAL_SCRIPTEXECUTE_PATH)/board/kernelconfig-scriptexecute.fragment"
BR2_LINUX_KERNEL_XZ=y
BR2_LINUX_KERNEL_DTS_SUPPORT=y
BR2_LINUX_KERNEL_INTREE_DTS_NAME="broadcom/bcm2711-rpi-5-b broadcom/bcm2712-rpi-cm5-cm5io overlays/dwc2-overlay overlays/spi-gpio40-45-overlay"
BR2_PACKAGE_E2FSPROGS=y
# BR2_PACKAGE_E2FSPROGS_FSCK is not set
BR2_PACKAGE_E2FSPROGS_RESIZE2FS=y
BR2_PACKAGE_FLASHROM=y
BR2_PACKAGE_PARTED=y
BR2_PACKAGE_RPI_USERLAND=y
BR2_PACKAGE_LIBARCHIVE=y
BR2_PACKAGE_LIBARCHIVE_BSDTAR=y
BR2_PACKAGE_ZLIB=y
BR2_PACKAGE_LIBSYSFS=y
BR2_PACKAGE_JITTERENTROPY_LIBRARY=y
BR2_PACKAGE_LIBCURL=y
BR2_PACKAGE_LIBCURL_CURL=y
# BR2_PACKAGE_LIBCURL_PROXY_SUPPORT is not set
# BR2_PACKAGE_LIBCURL_COOKIES_SUPPORT is not set
# BR2_PACKAGE_LIBCURL_EXTRA_PROTOCOLS_FEATURES is not set
BR2_PACKAGE_ARGP_STANDALONE=y
BR2_PACKAGE_UTIL_LINUX_BINARIES=y
BR2_TARGET_ROOTFS_CPIO=y
BR2_TARGET_ROOTFS_CPIO_XZ=y
# BR2_TARGET_ROOTFS_TAR is not set`

7. Executed 'make -C buildroot-2025.02 BR2_EXTERNAL="$PWD/scriptexecute" savedefconfig'
8. Executed 'build.sh' -> Failed for Firmware
9. Added the following to 'scriptexecutor/buildroot-2025.02/.config'
`BR2_PACKAGE_RPI_FIRMWARE_CUSTOM=y
BR2_PACKAGE_RPI_FIRMWARE_CUSTOM_DEFAULT=y
BR2_PACKAGE_RPI_FIRMWARE_CUSTOM_BOOT=""`

Dude90, 2025-04-15:
Short Update: By changing to a HDMI-HDMI cable (from a HDMI-DP cable) I was able to get the following display output. Directly after the content in the picture is displayed, the screen becomes black.

[image]

Dude90, 2025-04-17:
In the meantime I was finally successful provision a CM5. I am currently cleaning up all my files and will provide them here afterwards in some form (e.g. pull request, forked,...).

What I was not able to put into operation is the "recovery" of an CM5, e.g. setting a USB cable connection and a jumper to disable eMMC boot and by doing this trigger a new provision of a SOM that already has been provisioned. For the CM4 this always worked fine was kind of handy.
@maxnet Can you explain to me in more detail how this was realized? In the manual it states:
>This will cause the Compute Module to perform a USB boot, in which case the provisioning server will transfer the files of the utility OS over USB.
>After the utility OS has booted, it will contact the provisioning server over Ethernet to receive further instructions, and download additional files (e.g. the OS image to be written to eMMC) as usual.

In which mode does `rpiboot` start the SOM, `mass-storage` or `recovery`? If in `mass-storage` I don't get when and where the utility OS is copied to the SOM? Is in `recovery` I would assume the boot order to be somehow manipulated in order to perform a network boot even if the eMMC holds a valid image? I would be very happy if you could shed some light on this.

Thank you!

Dude90, 2025-04-23:
I am done sorting my files. This is the procedure which I followed to create a provision station for CM5 SOM:

1. Setup a normal cmprovision system, using the files and documentation in this repository.
2. Build a new scriptexecute OS using the following source: https://github.com/Dude90/scriptexecutor/tree/cm5_support. Just clone it and execute `build.sh` (not as root).
3. Copy the output files from step 2 (the files can be find in the folder `output`) to `var/lib/cmprovision/scriptexecute` on your cmprovision system from step 1 (replace or delete old files in this folder).
4. On your cmprovision system copy the file `/boot/overlay/bcm2712d0.dtbo` to `var/lib/cmprovision/scriptexecute/overlays`. Hint: Even though this file seems to be intended for 32bit systems it is somehow required for CM5 provision (which is using 64bit).
5. On your cmprovision system add the CM5 GUID by pasting `dhcp-match=set:client_is_a_pi,97,0:35:69:50:52` in `/var/lib/cmprovision/etc/dnsmasq.conf`.
6. If you want to get working final temperature of the SOMs after provision, change line 95 and 96 in `/var/lib/cmprovision/resources/views/scriptexecute.blade.php` as follows:
```
95   TEMP=`cat /sys/class/thermal/thermal_zone0/temp`
96   …-g "http://{{ $server }}/scriptexecute?serial={{ $cm->serial }}&alldone=1&temp=$((TEMP/1000))&verify={{ $project->verify }}"
```

Done! After this steps I have been able to use the provision station for CM5 SOMs.

Only the recovering of an already programmed SOM using a USB connection and the rpi_boot jumper is not working (see my last post). I would still be happy to get some feedback here.

I can give no guarantee that everything is working properly using the described setup but I hope this maybe helps somebody!

maxnet, 2025-04-24:
>@maxnet Can you explain to me in more detail how this was realized? In the manual it states

Cannot give much guidance with that. Do not have a CM5, so do not know if it has any quirks in regards to USB booting.
(My day job is not in the embeddded field, but more devops/site reliability engineering related. And do not need compute modules for hobby).

It just works by letting rpiboot feed the same Linux kernel and initramfs (containing the scriptexecute utility operating system) that is used when network booting over USB.

So do would expect that if booting those files by network boot works, it would also workover USB, provided right jumper is set (or button is pressed on normal Pi 5).
But again have not tried it...

Dude90, 2025-04-24:
Thank you for the explanation. I wasn't aware that a USB boot is possible with rpiboot by simply giving it a directory with the boot files.

I tried it again but I always run into the following error, independent of the `bootcode5.bin` I put in the `/var/lib/cmprovision/scriptexecute` folder:

>Apr 24 13:08:47 cm5provision2 rpiboot[391]: Waiting for BCM2835/6/7/2711/2712...
>Apr 24 13:08:47 cm5provision2 rpiboot[391]: Loading: /var/lib/cmprovision/scriptexecute/bootcode5.bin
>Apr 24 13:08:49 cm5provision2 rpiboot[391]: Sending bootcode.bin
>Apr 24 13:08:50 cm5provision2 rpiboot[391]: Failed control transfer (-7,24)
>Apr 24 13:08:50 cm5provision2 rpiboot[391]: Failed to write correct length, returned -7 

I also tried the current `bootcode5.bin` coming with the `rpiboot` tool, same result.
Maybe there is still something wrong with my `bootcode5.bin` or there is a general problem with `rpiboot` and the CM5. As many users had reported similar issues in the past (e.g. [here](https://github.com/raspberrypi/usbboot/issues/87)) I would guess the latter is the case.

I think I will for now just wait for an update of `rpiboot` and then try again. Hoping that the recovery function will eventually just start work again 😄

jayjayseal, 2025-04-25:
It is possible to boot and provision a pi5. But you will have to build a new kernel using buildroot-2024.08.1 along with a few other patches. I have made a new directory in the scriptexecute folder for when a pi5 is detected to use that kernel for example. Since the PI5 has new hardware several things have to be adjusted. In the blade things have to be changed as well.... (From the temperature readings (does not work at all in the current state), all the way up to the status red/green light when it is done provisioning since the path has changed). Since it's quite the change it will take me a while to list them all.

maxnet, 2025-04-25:
>This sometimes failed (probably due to a network blip)

May also happen if your image has highly compressible data, e.g. a lot of empty space/zeroes.
Those zeroes take very little space in the compressed image it downloads from the server, but writing them to SD card do can take long if it is say a gigabyte worth.
If it is busy writing those zeroes to SD card for more than one minute, and it downloads nothing new during that time, nginx closes the connection because it thinks download has stalled.

Can raise the timeout in the nginx config to see if that makes a difference.

E.g.:

on the cmprovision server, edit /etc/nginx/sites-available/cmprovision and toss in the "server" section:

```
send_timeout 600s;
```

Run "sudo systemctl restart nginx" afterwards.

jcaptitude, 2025-04-30:
Late response: I switched to rpi-sb-provisioner early this year and it's working fine.

I also tired @Dude90 's new instruction and it also works well! (Thank you for your investigation)

The new problem is when I tried to add eeprom file it always suggests :
`linux_spi_init: failed to open /dev/spidev0.0: No such file or directory`

I have turned on the SPI in raspi-config, so not sure why. I will try again later