raspberrypi/cmprovision #62: Custom Command Label Printing

SecareLupus, 2025-01-13:
Hello, I'm looking for any documentation on the label printing system. I am trying to get a Brother ptouch label printer working, somewhat similar to Issue #56, though I'm taking a different tack on it.

I'm using ptouch-print to do the actual printing on a remote device running an HTTP server to accept curl requests. I set the command for the custom command to a curl request against my service, and got test labels to print. I don't need the two fields below, providing a template and file extension, but it seems like the template is the only place where variable interpolation works. Is it possible to interpolate the $variables when used in the "Custom Command" field of a label printer?

The template field could even be helpful in this use case if it could be passed inline to the custom command, a la `curl http://XXXX:1234/print -d $template`

maxnet, 2025-01-13:
>The template field could even be helpful in this use case if it could be passed inline to the 
>custom command, a la curl http://XXXX:1234/print -d $template

Why inline? 
Can just tell curl to read from file with the "@" prefix.

```
curl -X POST -H "Content-Type: text/plain" -d @$file http://myserver/print
```

Then again, if you are doing web development anyway, you may as well just hook into the CmProvision code without the extra webserver in between.
E.g. listen for CmProvisioningComplete: https://github.com/raspberrypi/cmprovision/issues/45#issuecomment-1973076346

SecareLupus, 2025-01-14:
Holy cow, that `@$file` solution works perfectly. And that link to the other ticket will also likely come in handy. Thank you! Closing as this solved my issue!

closed 2025-01-14 by SecareLupus